StanfordVL/BEHAVIOR-1K #75: VR Support?

yukw777, 2022-12-21:
Do you have any plans for supporting VR in BEHAVIOR-1K as in BEHAVIOR-100? It seems like Omniverse has something called VCR (https://developer.nvidia.com/vcr-early-access) to record VR footage, though it's still in early access. Just wanted to hear your thoughts on this matter. Thank you for your help in advance!

cgokmen, 2022-12-21:
Hi Peter,

A VR setup just like BEHAVIOR is in the works right now and we expect to release it sometime early next year. Instead of Omniverse's native VR interfaces it would once again be a custom interface part of OmniGibson such that the interface also generates a valid sequence of robot actions that can be replayed via the Environment interface. We are trying to iron out some final performance issues prior to releasing this.

Thanks!

yukw777, 2022-12-23:
@cgokmen Amazing! Thanks for your reply. Very excited for it. Let me know if you want me to close this issue in the meantime.

cgokmen, 2022-12-24:
We can keep it open as a tracking issue. We might also have a branch for you to test out later. For now I'm assigning this to my teammate working on this.

closed 2024-01-13 by wensi-ai