StanfordVL/BEHAVIOR-1K #78: g_dataset missing

JiaheXu, 2022-12-28:
Hi OmniGibson developers,

    It looks like I am missing a dataset called "g_dataset", I was trying the scene_selector demo with `python -m omnigibson.examples.scenes.scene_selector`. I have tried to run the run_docker_gui.sh multiple times, but I still don't see it.

cremebrule, 2023-01-04:
Hi @JiaheXu ,

Thanks for bringing this up! The `g_dataset` is a legacy dataset of static scenes defined by a single mesh. Currently, we are not supporting those scenes, those these will be (re-)added shortly. I've gone ahead and updated the example so the static scenes cannot be selected. It will be updated again once we have those scenes ready for usage. Thanks!

closed 2023-01-04 by cremebrule

cgokmen, 2023-01-04:
I'll push an updated Docker image later today.