StanfordVL/BEHAVIOR-1K #79: Enable GPU dynamics flag issue

JiaheXu, 2022-12-28:
Hi OmniGibson developers,

    I am using the docker version OmniGibson. When I try 
`python -m omnigibson.examples.environments.behavior_env_demo`
it always shows
`2022-12-28 15:39:56 [559,151ms] [Error] [omni.physx.plugin] Particles feature is only supported on GPU. Please enable GPU dynamics flag in Property/Scene of physics scene!` error,  How can I enable the flag in OmniGibson env? Please help.

cremebrule, 2023-01-04:
Hi @JiaheXu ,

Thanks for bringing this up! Do you run into explicit physics errors or bad behavior?

Normally, the `Failed to find articulation at...` printout is expected, and doesn't necessarily mean anything terrible is happening. (In fact, we'd like to suppress this output sometimes so that users like yourself don't mistake it for an actual error message, but we haven't had time to look into this).

ChengshuLi, 2023-03-03:
Feel free to re-open this issue if you have any questions!

closed 2023-03-03 by ChengshuLi