StanfordVL/BEHAVIOR-1K #80: docker code Version

JiaheXu, 2022-12-28:
Hi OmniGibson developers,

    I have tried all the demos in the docker env, and none of them can work. The Isaac sim windows do show up, but they will automatically close due to memory issues. I have 64GB memory and a 3080Ti GPU, AMD 5950 CPU. 
[https://paste.ubuntu.com/p/xKM7rZGypy/](url) here is the detailed output, I think it might be a code version issue in docker env, please take a look, thanks.

JiaheXu, 2022-12-28:
My fault, the code just takes a lot of time to load all the models

ChengshuLi, 2023-01-03:
Hi @JiaheXu, these warning messages are caused by missing some metadata information for the Openable state for the fridge assets. You can safely ignore these messages for now, as they shouldn't break any example. We will fix the missing metadata in the next release soon. Thanks!

closed 2023-01-03 by ChengshuLi

JiaheXu, 2023-01-04:
Can you please tell me which are the most stable envs/settings to use for now? Thanks