StanfordVL/BEHAVIOR-1K #86: multi-agent support

ishikasingh, 2023-01-16:
Thanks for this really cool work!
I wanted to ask if it supports multi-agent? As in, can I instantiate multiple robot models in the same environment and have them execute actions in parallel - navigate, interact with objects?
If yes, can the agents interact with the same object at a particular time step? For instance, the case where the agent needs to hand off an object to another agent.

cgokmen, 2023-01-17:
Yes, we do support multi-robot use cases, and yes - it should be possible for multiple robots to handle the same object. The environment should accept either a dict or a concatenated list as the action for multiple robots, and produce a dict as the observation. Let us know if you run into any issues with this!

closed 2023-01-17 by cgokmen

ishikasingh, 2023-01-17:
Awesome! Thanks! Will let you know.

lhy0807, 2023-12-05:
Hi! It seems that each Task (eg. PointNavigationTask) is binding to only one robot. I wonder can we still achieve multiple agents training in such task?

hutslib, 2024-07-26:
@cgokmen hi! Is there any example or tutorial to run multi-robot?