StanfordVL/BEHAVIOR-1K #111: Error loading scene after fresh docker installation

jiaming-robot-learning, 2023-03-03:
Hi,

I followed the instructions and successfully installed it using docker. However, when I run it I got the following error:

```
Traceback (most recent call last):
  File "/micromamba/envs/omnigibson/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/micromamba/envs/omnigibson/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/omnigibson/omnigibson/examples/scenes/scene_selector.py", line 81, in <module>
    main()
  File "/omnigibson/omnigibson/examples/scenes/scene_selector.py", line 58, in main
    env = og.Environment(configs=cfg)
  File "/omnigibson/omnigibson/utils/python_utils.py", line 89, in wrapper
    func(*values.args, **values.kwargs)
  File "/omnigibson/omnigibson/envs/env_base.py", line 70, in __init__
    self.load()
  File "/omnigibson/omnigibson/envs/env_base.py", line 276, in load
    self._load_scene()
  File "/omnigibson/omnigibson/envs/env_base.py", line 198, in _load_scene
    og.sim.import_scene(scene)
  File "/omnigibson/omnigibson/simulator.py", line 300, in import_scene
    self.play()
  File "/omnigibson/omnigibson/simulator.py", line 506, in play
    obj.initialize()
  File "/omnigibson/omnigibson/prims/prim_base.py", line 79, in initialize
    self._initialize()
  File "/omnigibson/omnigibson/objects/dataset_object.py", line 206, in _initialize
    super()._initialize()
  File "/omnigibson/omnigibson/objects/stateful_object.py", line 142, in _initialize
    state.initialize(self._simulator)
  File "/omnigibson/omnigibson/object_states/object_state_base.py", line 89, in initialize
    self.reset()
AttributeError: 'InsideRoomTypes' object has no attribute 'reset'
Exception ignored in: <function _make_registry.<locals>._Registry.__del__ at 0x7f999d865ef0>
Traceback (most recent call last):
  File "/isaac-sim/kit/extscore/omni.kit.viewport.registry/omni/kit/viewport/registry/registry.py", line 103, in __del__
  File "/isaac-sim/kit/extscore/omni.kit.viewport.registry/omni/kit/viewport/registry/registry.py", line 98, in destroy
TypeError: 'NoneType' object is not callable
Exception ignored in: <function _make_registry.<locals>._Registry.__del__ at 0x7f999d865ef0>
Traceback (most recent call last):
  File "/isaac-sim/kit/extscore/omni.kit.viewport.registry/omni/kit/viewport/registry/registry.py", line 103, in __del__
  File "/isaac-sim/kit/extscore/omni.kit.viewport.registry/omni/kit/viewport/registry/registry.py", line 98, in destroy
TypeError: 'NoneType' object is not callable
Exception ignored in: <function SettingChangeSubscription.__del__ at 0x7f9be7f343b0>
Traceback (most recent call last):
  File "/isaac-sim/kit/kernel/py/omni/kit/app/_impl/__init__.py", line 114, in __del__
AttributeError: 'NoneType' object has no attribute 'get_settings'
Exception ignored in: <function RegisteredActions.__del__ at 0x7f8cf3a6e3b0>
Traceback (most recent call last):
  File "/isaac-sim/extscache/omni.kit.viewport.menubar.lighting-104.0.8/omni/kit/viewport/menubar/lighting/actions.py", line 345, in __del__
  File "/isaac-sim/extscache/omni.kit.viewport.menubar.lighting-104.0.8/omni/kit/viewport/menubar/lighting/actions.py", line 350, in destroy
TypeError: 'NoneType' object is not callable
2023-03-03 06:44:59 [39,452ms] [Warning] [carb.audio.context] 1 contexts were leaked
```

Any suggestions on this?

Thanks!

ChengshuLi, 2023-03-03:
You are right. Would you mind commenting out the following line for now? We will fix this issue in our next release very soon. Thanks for letting us know!

https://github.com/StanfordVL/OmniGibson/blob/915add3025d8d0dcf72239395c5c386f8548aa64/omnigibson/object_states/object_state_base.py#L89

closed 2023-03-07 by jiaming-robot-learning

cgokmen, 2023-03-17:
@nexthybrid include this in your entrypoint script right after `import omnigibson`

```
from omnigibson.macros import gm
gm.USE_GPU_DYNAMICS = True
```

ChengshuLi, 2023-03-17:
Just to clarify, if you don't need to use Fluid or Cloth, you can safely ignore this error.

nexthybrid, 2023-03-18:
> Just to clarify, if you don't need to use Fluid or Cloth, you can safely ignore this error.

@ChengshuLi  Thank you for the heads up. Then it looks like my root cause is something else. My current status: in a local build, my Isaac Sim did pop up, but then shut itself down briefly afterwards (lasted about 1 minute). Do you mind looking at my long of running this example on my local build (from the source repo)?

I activated the conda environment, then typed:
```
python -m omnigibson.examples.scenes.scene_selector
```
For the options I chose 1 for all three of them. Then Issac Sim started and seemed to be loading assets. Then after about a minute, it just quit. Here's a [link](https://gist.github.com/nexthybrid/f4a88337a7a127545fd9ecb5c5726815) to my entire terminal log.

ChengshuLi, 2023-03-18:
Did you see any thing on the Isaac Sim GUI? 

The source code is here: https://github.com/StanfordVL/OmniGibson/blob/main/omnigibson/examples/scenes/scene_selector.py

You should be able to see a TurtleBot being sampled at different locations executing random actions in an empty scene (with only walls, floors, and ceilings) if you chose 1 for all three options. The program will terminate in a bit or so after 10 episodes.