StanfordVL/BEHAVIOR-1K #112: object_state_base.py cannot initialize correctly

WangYixuan12, 2023-03-03:
Hi! First I would like to thank you for bringing this exciting project to the community. I can't wait to use it in my research. When I went through the tutorial, I found that running `python -m omnigibson.examples.scenes.scene_selector ` keeps failing here: https://github.com/StanfordVL/OmniGibson/blob/915add3025d8d0dcf72239395c5c386f8548aa64/omnigibson/object_states/object_state_base.py#L89. It seems that `reset` function is lost at this pull request: https://github.com/StanfordVL/OmniGibson/commit/b28c28cc1c563bb6e77207ad39dc5126886a543a#diff-c74f2de4c6fe927a4fec4f77a24709df614e5e73528dac31d137728c0d856d33L66. I am wondering whether my understanding is correct. If yes, could you please fix this problem? Thanks a lot for your help!

ChengshuLi, 2023-03-03:
You are right. Would you mind commenting out the following line for now? We will fix this issue in our next release very soon. Thanks for letting us know!

https://github.com/StanfordVL/OmniGibson/blob/915add3025d8d0dcf72239395c5c386f8548aa64/omnigibson/object_states/object_state_base.py#L89

WangYixuan12, 2023-03-03:
Thanks! It works for me.

closed 2023-03-03 by WangYixuan12