StanfordVL/BEHAVIOR-1K #114: AttributeError: 'HorizontalAdjacency' object has no attribute 'reset'

jiaming-robot-learning, 2023-03-03:
Hi,

I came across this error when I run the example after a fresh installation from source:

...
WARNING:root:Instance identifier 'root_handle' should be unique for adding to this registry mapping! Existing root_handle: None
2023-03-03 12:31:54 [35,904ms] [Error] [omni.physx.plugin] Particles feature is only supported on GPU. Please enable GPU dynamics flag in Property/Scene of physics scene!
2023-03-03 12:31:54 [35,904ms] [Error] [omni.physx.plugin] Particles feature is only supported on GPU. Please enable GPU dynamics flag in Property/Scene of physics scene!
2023-03-03 12:31:54 [35,904ms] [Error] [omni.physx.plugin] Particles feature is only supported on GPU. Please enable GPU dynamics flag in Property/Scene of physics scene!
2023-03-03 12:31:54 [35,904ms] [Error] [omni.physx.plugin] Particles feature is only supported on GPU. Please enable GPU dynamics flag in Property/Scene of physics scene!
2023-03-03 12:31:54 [35,951ms] [Warning] [omni.isaac.dynamic_control.plugin] Failed to find articulation at '/World/DustSystem/Dust_template'
2023-03-03 12:31:54 [35,952ms] [Warning] [omni.isaac.dynamic_control.plugin] Failed to find articulation at '/World/StainSystem/Stain_template'
2023-03-03 12:31:54 [35,952ms] [Warning] [omni.isaac.dynamic_control.plugin] Failed to find articulation at '/World/ceilings_ayfjzo_0'
Traceback (most recent call last):
  File "/home/user/anaconda3/envs/omnigibson/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/user/anaconda3/envs/omnigibson/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/data/omniverse/lib/isaac_sim-2022.2.0/OmniGibson/omnigibson/examples/scenes/scene_selector.py", line 81, in <module>
    main()
  File "/data/omniverse/lib/isaac_sim-2022.2.0/OmniGibson/omnigibson/examples/scenes/scene_selector.py", line 58, in main
    env = og.Environment(configs=cfg)
  File "/data/omniverse/lib/isaac_sim-2022.2.0/OmniGibson/omnigibson/utils/python_utils.py", line 89, in wrapper
    func(*values.args, **values.kwargs)
  File "/data/omniverse/lib/isaac_sim-2022.2.0/OmniGibson/omnigibson/envs/env_base.py", line 70, in __init__
    self.load()
  File "/data/omniverse/lib/isaac_sim-2022.2.0/OmniGibson/omnigibson/envs/env_base.py", line 276, in load
    self._load_scene()
  File "/data/omniverse/lib/isaac_sim-2022.2.0/OmniGibson/omnigibson/envs/env_base.py", line 198, in _load_scene
    og.sim.import_scene(scene)
  File "/data/omniverse/lib/isaac_sim-2022.2.0/OmniGibson/omnigibson/simulator.py", line 300, in import_scene
    self.play()
  File "/data/omniverse/lib/isaac_sim-2022.2.0/OmniGibson/omnigibson/simulator.py", line 506, in play
    obj.initialize()
  File "/data/omniverse/lib/isaac_sim-2022.2.0/OmniGibson/omnigibson/prims/prim_base.py", line 79, in initialize
    self._initialize()
  File "/data/omniverse/lib/isaac_sim-2022.2.0/OmniGibson/omnigibson/objects/dataset_object.py", line 206, in _initialize
    super()._initialize()
  File "/data/omniverse/lib/isaac_sim-2022.2.0/OmniGibson/omnigibson/objects/stateful_object.py", line 142, in _initialize
    state.initialize(self._simulator)
  File "/data/omniverse/lib/isaac_sim-2022.2.0/OmniGibson/omnigibson/object_states/object_state_base.py", line 89, in initialize
    self.reset()
AttributeError: 'HorizontalAdjacency' object has no attribute 'reset'
Exception ignored in: <function _make_registry.<locals>._Registry.__del__ at 0x7f76f66dab00>
Traceback (most recent call last):
  File "/data/omniverse/lib/isaac_sim-2022.2.0/kit/extscore/omni.kit.viewport.registry/omni/kit/viewport/registry/registry.py", line 103, in __del__
  File "/data/omniverse/lib/isaac_sim-2022.2.0/kit/extscore/omni.kit.viewport.registry/omni/kit/viewport/registry/registry.py", line 98, in destroy
TypeError: 'NoneType' object is not callable
Exception ignored in: <function _make_registry.<locals>._Registry.__del__ at 0x7f76f66dab00>
Traceback (most recent call last):
  File "/data/omniverse/lib/isaac_sim-2022.2.0/kit/extscore/omni.kit.viewport.registry/omni/kit/viewport/registry/registry.py", line 103, in __del__
  File "/data/omniverse/lib/isaac_sim-2022.2.0/kit/extscore/omni.kit.viewport.registry/omni/kit/viewport/registry/registry.py", line 98, in destroy
TypeError: 'NoneType' object is not callable
Exception ignored in: <function SettingChangeSubscription.__del__ at 0x7f78f4b95f80>
Traceback (most recent call last):
  File "/data/omniverse/lib/isaac_sim-2022.2.0/kit/kernel/py/omni/kit/app/_impl/__init__.py", line 114, in __del__
AttributeError: 'NoneType' object has no attribute 'get_settings'
Exception ignored in: <function RegisteredActions.__del__ at 0x7f6a549f6e60>
Traceback (most recent call last):
  File "/data/omniverse/lib/isaac_sim-2022.2.0/extscache/omni.kit.viewport.menubar.lighting-104.0.8/omni/kit/viewport/menubar/lighting/actions.py", line 345, in __del__
  File "/data/omniverse/lib/isaac_sim-2022.2.0/extscache/omni.kit.viewport.menubar.lighting-104.0.8/omni/kit/viewport/menubar/lighting/actions.py", line 350, in destroy
TypeError: 'NoneType' object is not callable
2023-03-03 12:31:55 [36,421ms] [Warning] [carb.audio.context] 1 contexts were leaked

```

I got the similar errors when install using the docker. It seems part of the code is buggy. Any ideas on this? Thanks!

ChengshuLi, 2023-03-03:
You are right. Would you mind commenting out the following line for now? We will fix this issue in our next release very soon. Thanks for letting us know!

https://github.com/StanfordVL/OmniGibson/blob/915add3025d8d0dcf72239395c5c386f8548aa64/omnigibson/object_states/object_state_base.py#L89

jiaming-robot-learning, 2023-03-07:
It solves this problem. Thanks!

closed 2023-03-07 by jiaming-robot-learning