StanfordVL/BEHAVIOR-1K #116: Human animation?

youngwoo-yoon, 2023-03-06:
Hello, 
Thanks for the great work.

I found the recent Omniverse has integrated human animation into scenes. 
(see https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/ext_omni_anim_people.html)
Do you have the plan to integrate human animation into OmniGibson?

ChengshuLi, 2023-03-06:
Hi [youngwoo-yoon](https://github.com/youngwoo-yoon), thanks for sharing with us the human animation extension from Omniverse. We don't have an immediate plan to integrate human animation into OmniGibson, but we receive feedback from many researchers who are interested in having humans in OmniGivson, so we might be able to incorporate this feature in the second half of this year.

youngwoo-yoon, 2023-03-06:
Thanks, looking forward to that!

closed 2023-03-06 by youngwoo-yoon