StanfordVL/BEHAVIOR-1K #129: og_dataset is missing average_category_spec.json

mjlbach, 2023-03-15:


nexthybrid, 2023-03-17:
I have a similar error when building from source then executing the following:
```
python -m omnigibson.examples.robots.robot_control_example
```
and after choosing the scene as the default scene:
```
Choose a robot (enter a number from 1 to 6): 1
INFO:root:Scene name: Rs_int
Traceback (most recent call last):
  File "/home/tong/miniconda3/envs/omnigibson/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/tong/miniconda3/envs/omnigibson/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/tong/pm/OmniGibson/omnigibson/examples/robots/robot_control_example.py", line 145, in <module>
    main()
  File "/home/tong/pm/OmniGibson/omnigibson/examples/robots/robot_control_example.py", line 95, in main
    env = og.Environment(configs=cfg, action_timestep=1/60., physics_timestep=1/60.)
  File "/home/tong/pm/OmniGibson/omnigibson/utils/python_utils.py", line 89, in wrapper
    func(*values.args, **values.kwargs)
  File "/home/tong/pm/OmniGibson/omnigibson/envs/env_base.py", line 70, in __init__
    self.load()
  File "/home/tong/pm/OmniGibson/omnigibson/envs/env_base.py", line 276, in load
    self._load_scene()
  File "/home/tong/pm/OmniGibson/omnigibson/envs/env_base.py", line 196, in _load_scene
    cls_type_descriptor="scene",
  File "/home/tong/pm/OmniGibson/omnigibson/utils/python_utils.py", line 274, in create_class_from_registry_and_config
    return cls(**cls_kwargs)
  File "/home/tong/pm/OmniGibson/omnigibson/utils/python_utils.py", line 89, in wrapper
    func(*values.args, **values.kwargs)
  File "/home/tong/pm/OmniGibson/omnigibson/scenes/interactive_traversable_scene.py", line 58, in __init__
    self.scene_dir = get_og_scene_path(scene_model)
  File "/home/tong/pm/OmniGibson/omnigibson/utils/asset_utils.py", line 112, in get_og_scene_path
    assert scene_name in os.listdir(og_scenes_path), "Scene {} does not exist".format(scene_name)
FileNotFoundError: [Errno 2] No such file or directory: '/home/tong/pm/OmniGibson/omnigibson/data/og_dataset/scenes'
```
Are we supposed to manually download this dataset into the repo from somewhere?

cgokmen, 2023-03-17:
@nexthybrid This is not a similar error - you just don't have the dataset installed. You can call omnigibson/scripts/setup.py to download the dataset, assets and encryption key.

nexthybrid, 2023-03-18:
> @nexthybrid This is not a similar error - you just don't have the dataset installed. You can call omnigibson/scripts/setup.py to download the dataset, assets and encryption key.

Thank you Cem @cgokmen, indeed I was not aware of the missing dataset. Now with your suggestion, I was able to run this robot_control_example.

ConnerQiu, 2023-03-21:
Hi @cgokmen , I come across the same error, it tells that: 
 ```shell
FileNotFoundError: [Errno 2] No such file or directory: '/home/cq/projects/SocialNav/OmniGibson/omnigibson/data/og_dataset/scenes'
2023-03-21 04:14:35 [9,783ms] [Warning] [carb.audio.context] 1 contexts were leaked
``` 

**While I run this setup.py, I receive the below error:** 
```shell
2023-03-21 06:38:03 [7,896ms] [Warning] [carb.flatcache.plugin] UsdRelationship /Render/RenderProduct_omni_kit_widget_viewport_ViewportTexture_0.orderedVars has multiple targets, which is not supported

Welcome to OmniGibson!

Downloading assets...
Assets already downloaded.

OmniGibson setup completed!

2023-03-21 06:38:04 [8,538ms] [Warning] [carb.audio.context] 1 contexts were leaked
``` 

How to fix this leaked contexts problem

ChengshuLi, 2023-04-07:
`2023-03-21 06:38:04 [8,538ms] [Warning] [carb.audio.context] 1 contexts were leaked`
This error from omniverse can be safely ignored.

closed 2023-04-07 by ChengshuLi