StanfordVL/BEHAVIOR-1K #131: Add human agents in the simulator

ConnerQiu, 2023-03-16:
Hi, I would to like if OmniGibson can also add human agents in the simulator and also control their behavior, just like the iGibson. The paper show a an example: https://www.youtube.com/watch?v=aFwbK-hpyjk&t=176s.

wensi-ai, 2023-03-16:
Hi @ConnerQiu, yes OG currently supports multi-agent. You can specify more than one robot in the robot config, and for env.step(action), you will pass in an action dict. We will incorporate Behaviorbot (just like iGibson) in the near future where you can use this method to add and control multiple humanoid robot in a scene.

There is another [omniverse human animation extension](https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/ext_omni_anim_people.html) where you can add human character into scenes. We don't have an immediate plan to integrate this but might be able to do so in the second half of this year.

ConnerQiu, 2023-03-16:
Hi, @wensi-ai Thanks for your quick response! So, once the Behaviorbot is incorporated in OG, we can the current method of adding multiple robots to add multiple humanoid robots. Can we can add self-define robots? So that I can define some humanoid robots, and then add them in the environment. Or do you have a schedule about incorporating the Behaviorbot, since I plan to start my project in OG recently. Maybe there is some way we can corporate to avoid some redundant work.

wensi-ai, 2023-03-16:
We are finetune the Behaviorbot and will be releasing it next month. Meanwhile you can also add your self-defined robot into OG by writing a custom robot class that inherits BaseRobot. For that you can take a look at the robot classes under `omnigibson/robots` folder

closed 2023-03-20 by ConnerQiu