StanfordVL/BEHAVIOR-1K #132: Error when running examples with deformable bodies

Murkey8895, 2023-03-20:
Hi,
Thanks for your great work!

I came across this error when I run the examples with deformable boides (in folded_unfolded_state_demo.py and overlaid_demo.py):
`AttributeError: 'ClothPrim' object has no attribute 'kinematic_only'`

Would you please give any suggestions on this?

Thank you so much!

ChengshuLi, 2023-03-20:
Thanks for reporting this. We have fixed this issue and will include the fix in the next public release.

Meanwhile, please add the following lines of code the cloth_prim.py as a temporary walkaround. 

```
    @property
    def kinematic_only(self):
        """
        Returns:
            bool: Whether this object is a kinematic-only object. For ClothPrim, always return False.
        """
        return False
```

Let me know if this addresses this issue.

Murkey8895, 2023-03-21:
It solves this problem, but there is others function (`check_smoothness()` and `check_projection_area_and_diagonal()`) missing in `folded_state_demo.py` (in the tutorial):

`AttributeError: 'Folded' object has no attribute 'check_projection_area_and_diagonal'`

Thanks!

ChengshuLi, 2023-03-21:
Hmmm `folded_state_demo.py` is probably outdated in the tutorial. We will update it.  

For now, can you use `folded_unfolded_state_demo.py`? Does it work for you?
https://github.com/StanfordVL/OmniGibson/blob/main/omnigibson/examples/object_states/folded_unfolded_state_demo.py

Murkey8895, 2023-03-22:
It works, thank you again!

closed 2023-03-30 by ChengshuLi