StanfordVL/BEHAVIOR-1K #133: No module named 'omni' from install from source

ConnerQiu, 2023-03-21:
**I got this issue when build from source:**
INFO:root:Importing OmniGibson (omnigibson module)
INFO:root:Assets path: /home/cq/projects/SocialNav/OmniGibson/omnigibson/data/assets
INFO:root:Gibson Dataset path: /home/cq/projects/SocialNav/OmniGibson/omnigibson/data/g_dataset
INFO:root:OmniGibson Dataset path: /home/cq/projects/SocialNav/OmniGibson/omnigibson/data/og_dataset
INFO:root:OmniGibson Key path: /home/cq/projects/SocialNav/OmniGibson/omnigibson/data/omnigibson.key
INFO:root:Example path: /home/cq/projects/SocialNav/OmniGibson/omnigibson/examples
INFO:root:Example config path: /home/cq/projects/SocialNav/OmniGibson/omnigibson/configs
Traceback (most recent call last):
  File "/home/cq/miniconda3/envs/omnigibson/lib/python3.7/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/home/cq/miniconda3/envs/omnigibson/lib/python3.7/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/home/cq/projects/SocialNav/OmniGibson/omnigibson/__init__.py", line 144, in <module>
    REGISTERED_TASKS, ALL_SENSOR_MODALITIES = start()
  File "/home/cq/projects/SocialNav/OmniGibson/omnigibson/__init__.py", line 126, in start
    app = create_app()
  File "/home/cq/projects/SocialNav/OmniGibson/omnigibson/__init__.py", line 95, in create_app
    from omni.isaac.kit import SimulationApp
ModuleNotFoundError: No module named 'omni'

It seems the omni module is not included in the omnigibison package. Is there way to solve this, thanks!

wensi-ai, 2023-03-21:
Hi @ConnerQiu, could you try removing the omnigibson env and source the setup script by running `source setup_conda_env.sh`? Thanks!

ConnerQiu, 2023-03-21:
**Hi @wensi-ai Thanks! The previous problem seems to be solved! However, I still get a erroe message from the command line:**
ModuleNotFoundError: No module named 'omni.particle'
2023-03-21 03:32:20 [10,223ms] [Warning] [carb.audio.context] 1 contexts were leaked
Segmentation fault (core dumped)
**Here are some system info of my device:**
Warp 0.6.3 initialized:
   CUDA Toolkit: 11.5, Driver: 12.0
   Devices:
     "cpu"    | x86_64
     "cuda:0" | NVIDIA GeForce RTX 3070 Ti (sm_86)
   Kernel cache: /home/cq/.cache/warp/0.6.3
[3.070s] [ext: omni.syntheticdata-0.2.4] startup

I make sure that the examples are runned in the omnigibson envrionment.
Previously, I got an error msg about ray tracing, but somehow such msg just disappeared. A bit wired.

wensi-ai, 2023-03-21:
Hi @ConnerQiu, could you try the setup_conda_env.sh script under the `og-develop` branch? These installation problems are pending to be fixed in our next release which will happen later this month.

ConnerQiu, 2023-03-21:
hi @wensi-ai I am not very clear about the og-develop branch you said. Is it a new OmniGibson packge that is under development? If so, how can I get it. The newest version I can see is the V0.0.5.

wensi-ai, 2023-03-21:
You can replace the content in the setup script with [this one](https://github.com/StanfordVL/OmniGibson/blob/og-develop/setup_conda_env.sh) as a temporary solution for now. We will include this in our 0.0.6 release.

ConnerQiu, 2023-03-21:
well thanks! Now I get what you just said, it is on the top-left of the repo. I guess the errors in OG maybe be solved. But I still get an error :**Waiting for compilation of ray tracing shaders by GPU driver: 60 seconds so far**. I have seen this when I run omniverse alone. Do you know how to solve this. I guess this has to do with the omniverse, maybe I should find other informantion about this issue.

closed 2023-04-06 by ChengshuLi