StanfordVL/BEHAVIOR-1K #138: [Error] [omni.physx.plugin] Particles feature is only supported on GPU. Please enable GPU dynamics flag in Property/Scene of physics scene!

michalvavrecka, 2023-03-30:
I am sorry for the stupid question, but I do not know how to enable this feature as the omnigibson crashes during initialization followed by the AttributeError: 'ContactParticles' object has no attribute 'reset in terminal. Is this a bug or does it realate to GPU (RTX2060 card, 525 drivers)?

Thanks

Michal

wensi-ai, 2023-03-30:
could you try setting `gm.USE_GPU_DYNAMICS = True` in macros.py?

ChengshuLi, 2023-03-30:
Sorry it's a bug on our end. To resolve the issue of " 'ContactParticles' object has no attribute 'reset" would you mind commenting out the following line for now? We will fix this issue in our next release very soon. Thanks for letting us know!

https://github.com/StanfordVL/OmniGibson/blob/915add3025d8d0dcf72239395c5c386f8548aa64/omnigibson/object_states/object_state_base.py#L89

michalvavrecka, 2023-03-31:
Thanks, the changes in the master branch did not work, but when I switched to developmental branch, the errors disappeared. I was able to run basic demos with turtlebot, but rest of the robot still waits for the conversion from iGibson.

tx871217, 2023-04-02:
hi, sry dumb question. i used docker to setup OmniGibson instead of from source. any ideas to switch to dev branch? im facing the same issue. Thanks!

wensi-ai, 2023-04-04:
Hi @tx871217 , we currently only support main branch for docker install. We will have a new release this week which would update the main branch to og-dev.

closed 2023-04-17 by wensi-ai