StanfordVL/BEHAVIOR-1K #145: run Docker error

1412kauti, 2023-04-04:
```
kaito@comet:~/Desktop$ ./run_docker.sh 
Missing /usr/share/vulkan/icd.d/nvidia_layers.json file.
(default path: /usr/share/vulkan/icd.d/nvidia_layers.json)
In some distributions this file will be at /etc/vulkan/implicit_layer.d/
Consider updating your driver to 525 if you cannot find the file.
To continue update the file path at the top of the run_docker.sh file and retry
```
I am running this in PopOS 20.04 , ive installed nvidia-docker2 and am running the 525 nvidia driver as well...I was able to find the nvidia_layers.json file as well, but im getting this error

wensi-ai, 2023-04-04:
Hi @1412kauti , where is your `nvidia_layers.json` located? For some OS it could be at `/usr/share/vulkan/implicit_layer.d`. We will have a new release this week which hopefully could solve this issue.

1412kauti, 2023-04-06:
yup...got it working thanks....had to do

```
sudo apt install mlocate

locate nvidia_layers.json
locate implicit_layer.d
```

and update it accordingly