StanfordVL/BEHAVIOR-1K #149: Loading a custom USD

grimereaper7471, 2023-04-06:
{
        "type": "USDObject",
        "name": "Cube",
        "usd_path": "/home/kaito/Documents/Expt/OmniGibson/ZCode/USD_Folder/Cube.usd",
        "category": "Cube",
        "visual_only": True,
        "scale": [2.0, 1.0, 2.0],
        "position": [3.0, 0, 2.0],
        "orientation": [0, 0, 0, 10.0],
    },

I was trying to load a custom USD in the scene useing this code where the USD and it didn't work and I was wondering if there is a tutorial for loading custom USDs or am I stuck to just the dataset.

ChengshuLi, 2023-04-06:
Could you provide more information? What is the error that you encountered?

cgokmen, 2023-08-01:
Make sure you set USE_ENCRYPTED_DATASET=False when loading this object. You might need to modify the code a little bit to be able to load the combination of encrypted & unencrypted assets. I'll open us a tracking issue to implement better support for this.

cgokmen, 2023-08-01:
Actually the parameter I mentioned above is not currently available. We'll implement this. Thanks