StanfordVL/BEHAVIOR-1K #162: Object error in examples

ConnerQiu, 2023-04-11:
Great that the new version is released. The installation process is simple and clean!
when I run both of the examples secen_selector and robot_control, I got the same error meassage, it seems like relates to the method "self.disable_collisions_for_fixed_objects()" is used in wrong objects:

`[INFO] [omnigibson.objects.object_base] Loaded walls_xlgydu_0 at /World/walls_xlgydu_0
Traceback (most recent call last):
  File "/home/cq/miniconda3/envs/omnigibson/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/cq/miniconda3/envs/omnigibson/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/cq/projects/SocialNav/OmniGibson/omnigibson/examples/robots/robot_control_example.py", line 143, in <module>
    main()
  File "/home/cq/projects/SocialNav/OmniGibson/omnigibson/examples/robots/robot_control_example.py", line 93, in main
    env = og.Environment(configs=cfg, action_timestep=1/60., physics_timestep=1/60.)
  File "/home/cq/projects/SocialNav/OmniGibson/omnigibson/utils/python_utils.py", line 90, in wrapper
    func(*values.args, **values.kwargs)
  File "/home/cq/projects/SocialNav/OmniGibson/omnigibson/envs/env_base.py", line 81, in __init__
    self.load()
  File "/home/cq/projects/SocialNav/OmniGibson/omnigibson/envs/env_base.py", line 319, in load
    self._load_scene()
  File "/home/cq/projects/SocialNav/OmniGibson/omnigibson/envs/env_base.py", line 197, in _load_scene
    og.sim.import_scene(scene)
  File "/home/cq/projects/SocialNav/OmniGibson/omnigibson/simulator.py", line 316, in import_scene
    self._scene.load()
  File "/home/cq/projects/SocialNav/OmniGibson/omnigibson/scenes/scene_base.py", line 247, in load
    self._load_objects_from_scene_file()
  File "/home/cq/projects/SocialNav/OmniGibson/omnigibson/scenes/scene_base.py", line 209, in _load_objects_from_scene_file
    self.disable_collisions_for_fixed_objects()
AttributeError: 'InteractiveTraversableScene' object has no attribute 'disable_collisions_for_fixed_objects'
[INFO] [omnigibson] ---------- Shutting Down OmniGibson ----------
[33.371s] Simulation App Shutting Down
`

Before it shut down, I got serval warning, I list them all in sequence here( they happened in different places):
`[Warning] [omni.isaac.kit.simulation_app] Modules: ['omnigibson', 'omnigibson.macros', 'omni.isaac.kit.app_framework'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the following args:  ['/home/cq/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/cq/.local/share/ov/pkg/isaac_sim-2022.2.1/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/home/cq/.local/share/ov/pkg/isaac_sim-2022.2.1/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=False', '--/app/fastShutdown=True', '--ext-folder', '/home/cq/.local/share/ov/pkg/isaac_sim-2022.2.1/exts', '--ext-folder', '/home/cq/.local/share/ov/pkg/isaac_sim-2022.2.1/apps', '--/physics/cudaDevice=0', '--portable']
Passing the following args to the base kit application:  []
[Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.

2023-04-10 13:44:48 [13ms] [Warning] [omni.ext.plugin] [ext: omni.sensors.nv.lidar] Extensions config 'extension.toml' doesn't exist '/home/cq/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.sensors.nv.lidar' or '/home/cq/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.sensors.nv.lidar/config'
2023-04-10 13:44:48 [13ms] [Warning] [omni.ext.plugin] [ext: omni.sensors.nv.radar] Extensions config 'extension.toml' doesn't exist '/home/cq/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.sensors.nv.radar' or '/home/cq/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.sensors.nv.radar/config'

2023-04-10 13:44:49 [1,748ms] [Warning] [omni.kit.viewport.menubar.core.menu_item.viewport_menu_item] ViewportMenuItem order argument is deprecated, use order_setting_path

2023-04-10 13:44:50 [2,749ms] [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template

2023-04-10 13:44:50 [2,811ms] [Warning] [omni.ext.plugin] [ext: omni.isaac.sensor-5.6.3] Native plugin search dir contains several files: /home/cq/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.isaac.sensor/bin . It makes dir walk slow, replace with the exact path to the plugin instead of a wildcard.

2023-04-10 13:44:51 [3,376ms] [Warning] [omni.isaac.isaac_sensor] omni.isaac.isaac_sensor is deprecated and was renamed; use omni.isaac.sensor

2023-04-10 13:44:55 [7,425ms] [Warning] [carb.flatcache.plugin] UsdRelationship /Render/RenderProduct_omni_kit_widget_viewport_ViewportTexture_0.orderedVars has multiple targets, which is not supported
'
These information

wensi-ai, 2023-04-11:
Hi, we just discovered the problem today. This should be fixed in v0.1.0 (which we released today). Please use the new docker image or pull from the main branch if you are installation from source.

Isaac Sim does spill out a lot of warnings, so in general if those do not crash OG you can safely ignore them.

ConnerQiu, 2023-04-11:
Great. Highly efficient! Also thanks for you advice, I am considering switching to 2022.2.0 to avoid issues in the future~

closed 2023-04-11 by ConnerQiu