StanfordVL/BEHAVIOR-1K #179: Recommended workflow for parallelizing the environment

sai-prasanna, 2023-05-12:
Hi, what would be the recommended workflow/APIs for vectorizing the environment? I think Isaac based environments clone multiple instances of the task/agent/environment in the same simulator instance. The current example for RL training has only single environment, I was wondering if you have a workflow planned for this or already in place in your internal experiments.

cgokmen, 2023-08-01:
Isaac Sim environments do not do this as far as we know - to parallelize, you can launch multiple Docker containers containing multiple instances of OmniGibson.

closed 2023-08-01 by cgokmen

sai-prasanna, 2023-08-01:
Isaac gym spawns multiple robots in the same environment and exposes an API to support parallel actions etc. I think its possible to replicate the rooms, robots and all assets, translate it all and just changing the agent position returned for each robot by the translation where the "instance" is spawned.

IsaacGym already allows one to spawn multiple robots for their tasks https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs

Parallelizing with docker on the same GPU would be really inefficient way to parallelize as the processes using CUDA will be executed sequentially on the GPU, not to mention the overhead of transfering observations across processes.

cgokmen, 2023-08-01:
Isaac Gym and Isaac Sim are two different products by nvidia. Isaac Gym is aimed for reinforcement learning research & supports parallelization as you mention. This is not true for Isaac Sim currently, which OmniGibson is built on - and we are not aware of any good ways of parallelizing the environment besides running multiple instances.

cgokmen, 2023-08-01:
cc @cremebrule

sai-prasanna, 2023-08-01:
Isaac gym pretty much runs on isaac sim. And the way they have done the parallelization is pretty much spawning the elements of the scene repeatedly in multple places. For indoor environments in omnigibson, the walls already enclose the elements of the scene, so spawning all of it multiple times similar to isaac-gym might work!

Have a look at cloner https://docs.omniverse.nvidia.com/isaacsim/latest/tutorial_gym_cloner.html#getting-started

This allows us to do what I am suggesting.

cgokmen, 2023-08-01:
It is unfortunately not possible to do this with OmniGibson currently. Even if you are able to clone the underlying prim tree (which probably would work) none of the OmniGibson code is compatible with something like this and everything explicitly assumes a single copy of the scene. You could create a pseudo-scene containing multiple copies of the scene and place a robot in each copy, but it would not prevent task code from using objects cross-scene, etc.

It is however an interesting prospect - running multiple environments in a single Isaac Sim environment would allow reusing the GPU resources used for rendering. Regardless, much of the computationally inefficient work in OmniGibson happens in Python in the object state logic, and is impossible to parallelize easily.

cgokmen, 2024-04-05:
Currently, yes. We are working on this feature (many parallel environments in one OmniGibson process) and have a really exciting minimum working example ready and will likely be releasing it in the upcoming month, however.