StanfordVL/BEHAVIOR-1K #181: Code for RL-Prim training in the paper

qianLyu, 2023-05-16:
Hi authors,

Thanks for sharing your wonderful work! I'm wondering if there's any demo reimplementing the RL-Prim training (CollectTrash/CleanTable) shown in the paper?

Thanks in advance!

cgokmen, 2023-08-01:
This code is not currently released as part of OmniGibson. We are working on a separate set of action primitives that will be available as part of an upcoming release, probably in September.

Same as #238

closed 2023-08-01 by cgokmen

lziiid, 2024-05-09:
> This code is not currently released as part of OmniGibson. We are working on a separate set of action primitives that will be available as part of an upcoming release, probably in September.
> 
> Same as #238

Thanks for the wonderful work and open-source code! I wonder if the set of action primitives is available now? I can't find it in the source code :(