StanfordVL/BEHAVIOR-1K #187: Noise in observations or pose

sai-prasanna, 2023-05-26:
I am implementing 2D bird's eye semantic mapping for omnigibson. I use a library called [dungeon_maps](https://github.com/Ending2015a/dungeon_maps) which takes depth and semantic labels to construct the map. It uses camera parameters (hfov, resolution) and camera absolute pose to project the semantic pointcloud constructed from depth map onto a top down view. I am building a global map.

I run into a problem where when I move the robot around, the objects slightly move around. ie There is noise in either the pose or the observation. 

FYI, to get the absolute pose, I add the x,y coordinate of camera position to the robot position. One thing that's weird is that the same `get_position_orientation` api returns position/orientation w.r.t global frame for the robot, but I guess for the sensor it returns the position/orientation relative to the robot base link. I might be wrong here and missing some other transformation which might result in the noise.

Here is the brief code to replicate what I am talking about. Do a `pip install dungeon_maps` in whatever environment you're using.

``` python
import math
from typing import List, Tuple
import numpy as np

import omnigibson as og
from omnigibson.macros import gm
from omnigibson.robots import REGISTERED_ROBOTS
from omnigibson.object_states import Pose

from omnigibson.utils.ui_utils import choose_from_options, KeyboardRobotController
import dungeon_maps as dmap
from omnigibson.utils.usd_utils import get_camera_params
from omnigibson.utils.transform_utils import euler2quat, quat2euler, mat2euler, quat_multiply
import torch
from dungeon_maps.demos.object_map import vis
import cv2
from PIL import Image
CONTROL_MODES = dict(
    random="Use autonomous random actions (default)",
    teleop="Use keyboard control",
)

SCENES = dict(
    Rs_int="Realistic interactive home environment (default)",
    empty="Empty environment with no objects",
)

# Don't use GPU dynamics and use flatcache for performance boost
gm.USE_GPU_DYNAMICS = False
gm.ENABLE_FLATCACHE = True

def choose_controllers(robot, random_selection=False):
    """
    For a given robot, iterates over all components of the robot, and returns the requested controller type for each
    component.

    :param robot: BaseRobot, robot class from which to infer relevant valid controller options
    :param random_selection: bool, if the selection is random (for automatic demo execution). Default False

    :return dict: Mapping from individual robot component (e.g.: base, arm, etc.) to selected controller names
    """
    # Create new dict to store responses from user
    controller_choices = dict()

    # Grab the default controller config so we have the registry of all possible controller options
    default_config = robot._default_controller_config

    # Iterate over all components in robot
    for component, controller_options in default_config.items():
        # Select controller
        options = list(sorted(controller_options.keys()))
        choice = choose_from_options(
            options=options, name="{} controller".format(component), random_selection=random_selection
        )

        # Add to user responses
        controller_choices[component] = choice

    return controller_choices
from torch import nn

[... 206 lines cut ...]
    assert topdown_map.proj is not None
    pos = np.asarray(trajectory, dtype=np.float32)
    pos = topdown_map.get_coords(pos, is_global=True)
    pos = dmap.utils.to_numpy(pos)[0]
    return draw_segments(image, pos, color=color, size=size)

def draw_segments(image, points, color, size=2):
    for index in range(1, len(points)):
        prev = index - 1
        cur = index
        if np.all(points[prev] == points[cur]):
            continue
        image = draw_segment(image, points[prev], points[cur], color, size)
    return image

def draw_segment(image, p1, p2, color, size=2):
    image = cv2.line(image, (p1[0], p1[1]), (p2[0], p2[1]), color=color, thickness=size)
    return image

if __name__ == "__main__":
    main()

```

Visually, when the robot moves around the observations have this bounce (tilt) to it, but I don't use any sensor noises and I assume the coordinates returned will be the true coordinates.

cgokmen, 2023-08-01:
For the noise issue here, could you try disabling the denoising stuff Omniverse has (DLSS, etc.) - these should be in the render settings menu in the RTX Realtime Settings. https://docs.omniverse.nvidia.com/materials-and-rendering/latest/103/render-settings_overview.html

The visualsensor local pose thing seems to be a separate issue so I opened a new issue for that.

Also this dungeon_maps thing looks really interesting. Perhaps we can add direct support for it later.

sai-prasanna, 2023-08-14:
@ChengshuLi It should, but it does not. The value returned there is always the same regardless of where the robot is. Which is why I called it relative pos, ori. I guess that's why @cgokmen raised https://github.com/StanfordVL/OmniGibson/issues/243 as an issue.

cgokmen, 2023-08-14:
Let's try to verify this @ChengshuLi - I do agree there might actually be a bug here.

ChengshuLi, 2023-08-14:
Got it. Let me investigate a bit more and get back to you.

ChengshuLi, 2023-08-14:
@sai-prasanna Sorry another question, which robot are you using?

cremebrule, 2023-08-15:
Sorry, following up on this as well. I just tested and verified that **without** `gm.ENABLE_FLATCACHE` on, VisionSensors should return the correct value, since every prim's attributes (including pos / ori) gets updated every time an object moves in the sim.

However, `gm.ENABLE_FLATCACHE` (which in your attached script, is set to `True`) causes this to not work, which is expected because it is precisely the bypassing of writing to the USD at every timestep that gives the performance boost associated with flatcache.

Can you try setting `gm.ENABLE_FLATCACHE=False` and trying again?

ChengshuLi, 2023-08-15:
@cremebrule if @sai-prasanna is using an ActiveCameraRobot with the camera attachehd to a moving link, even with `gm.ENABLE_FLATCACHE=False`, the result will still be wrong, right?

sai-prasanna, 2023-08-22:
@cremebrule  Thanks! setting it makes the position and orientation return the values I was computing manually.

chenjiayi-zxx, 2023-11-20:
@cremebrule Hello, have you successfully used Fetch to build a map, can you share your method, I seem to be using Fetch when the result is even worse, thank you.

cgokmen, 2024-01-11:
Closing this as resolved, the concrete issue to fix can be found in #243

closed 2024-01-11 by cgokmen