StanfordVL/BEHAVIOR-1K #225: Tiago's arm motion is very slow when using action primitives

misoshiruseijin, 2023-07-21:
When using Semantic Action Primitives with Tiago robot, the arm motion is very slow, especially when moving the shoulder joints to lift the arm up. The Fetch robot does not have this problem. Increasing the Kp gain in the ControllableObject class (from default 1e7 to 1e35) only makes it marginally faster. Reducing the arm mass speeds up the arm significantly, but is there a cleaner fix?

cgokmen, 2023-08-01:
Any idea how this was resolved? I'm closing but feel free to leave a comment.

closed 2023-08-01 by cgokmen

misoshiruseijin, 2023-08-02:
I suspect it has to do with GPU and driver versions but I am not completely sure. The arm started moving a lot faster on one of the workstations after updating the graphics driver, but it is not as fast on other workstations.