StanfordVL/BEHAVIOR-1K #230: OmniGibson Python Version Compatibility Inquiry

GuanxingLu, 2023-07-26:
Hi there,

I've been using OmniGibson and noticed that it currently supports Python 3.7 strictly [ref: https://forums.developer.nvidia.com/t/using-my-own-anaconda-env-modulenotfounderror-no-module-named-carb-carb/242907]. I'm interested in utilizing some extended techniques that are available in Python 3.8 and higher versions.

I'd like to inquire whether there are any plans to update OmniGibson to be compatible with Python 3.8 or even higher versions. If there are no immediate plans, I'd appreciate any guidance on how to use OmniGibson with Python versions above 3.7.

Thank you for your work on this project and any assistance you can provide!

cgokmen, 2023-07-26:
Hi,

We were informed by Nvidia that Isaac Sim would be updated to 3.10 in the upcoming month or so. We will update to match that version once the new Isaac Sim version is out.

Thanks,
Cem

GuanxingLu, 2023-07-26:
Got it, thank you for your prompt response!

closed 2023-07-26 by GuanxingLu