StanfordVL/BEHAVIOR-1K #240: How to use omni.anim.people in omnigibson?

15210579725, 2023-08-01:
Hi, 
thanks for great work!

Is there any way to use extensions such as omni.anim.people before integrated in omnigibson?

cgokmen, 2023-08-01:
Hi,

We don't have any built-in support for this currently, but it looks like a really nice use case. You could try to do it yourself:

1. Launch OmniGibson with one of the scenes pre-loaded
2. Insert into your code, right before what you want to do with the environment, a loop like this to let the UI run:
```
try:
   while True: og.sim.step()
except InterruptError:
   pass
```
3. Set up the omni.anim.people extension using the instructions in this doc: https://docs.omniverse.nvidia.com/isaacsim/latest/ext_omni_anim_people.html
4. When done, CTRL+C to stop the loop, and your code should continue to w/e you have in the loop, and hopefully now the people are still in the scene.

Note that we have not tried this out (although we'd like to, at some point) so this may or may not work. Please let us know here if you find that it works or feel free to make PRs to integrate this as a feature!

Thanks,
Cem

15210579725, 2023-08-03:
Thanks a lot for your suggestion!
Seems like 'og.sim.render()' can't make anim.people move when I add a Tiago robot with two people into a empty scene
However, those people could move when I add them into a empty scene without a robot
I feel hopeful about it
Is there any low-level code that render anim.people one step?
Can I call the environment to render anim.people one step and then render omnigibson another step, alternating between two steps?
Looking forward to your reply!

[image]
[image]

cgokmen, 2023-08-03:
og.sim.render() just renders the UI, it doesn't actually step the simulation. It's the env.step that steps the simulation (this should also move your humans) but I'm not sure why it doesn't work. Can you try to find out, outside the context of OmniGibson, how to make the humans move via code? Did you do all the stuff recommended here: https://docs.omniverse.nvidia.com/isaacsim/latest/ext_omni_anim_people.html ? Did you try the approach in 3.5 and seeing what's in that python script?

15210579725, 2023-08-07:
Sorry for not replying you for 3 days.
It seems like the reason that "anim.people" can't move is because I have to "Load" and "Set" character through menu manually Only when Omniverse is "stopped". But I was trying to "Load" and "Set" character when Omniverse is "Paused" 2 days before, so it doesn't work. 
But when I click "Stop" button, after the program has set up Omnigibson, the robot will cause the program to crash and exit Omniverse. 
The first way to solve this is to manually start load character before the environment is set up and start to run, but I found insert "time.sleep()" before "env = og.Environment(cfg)" doesn't work, because I can't click the Omniverse environment when the program is sleeping. Is there any way to let me load the character before the environment start to run?
The second way might be when I click "stop", the robot will go back to its initial position instead of the program crashed and exit Omniverse. Do you have any suggestions?
I believe it will run very soon!

ChengshuLi, 2023-08-13:
Can you try the following?

```
env = og.Environment(...)
assert og.sim.is_playing()  # sim should be playing now
og.sim.stop()
assert og.sim.is_stopped()  # sim should be stopped now
for _ in range(1000):  # This will allow you to access the omniverse UI. If you need more time, re-run the for loop.
    og.sim.render()
[While the for loops is running, on omniverse's UI, load the people and do whatever setup you need]
og.sim.play()
assert og.sim.is_playing()  # sim should be playing now
og.sim.scene.reset()  # this should reset the scene to its initial state.

for _ in range(1000):
    og.sim.step()    # Now everything should run smoothly, hopefully, with the people moving around
```

Let me know if this works! 

If this works properly, you should properly add the people and set their motions using python interface rather than the omniverse UI of course. And we would be thrilled if you can file a PR to augment OG with this very nice feature!

15210579725, 2023-08-24:
Thanks a lot!
your advice works really well!
Now the robot and the human could move altogether.
But currently the human doesn't has collision size and is transparent to the robot
[image]

And omniverse doesn't provide delicate Python API to control anim.people besides from simple txt file added in the control panel
It's still hard to control those character doing behavior what I asked now.

Yuuraa, 2025-03-29:
Hi, thank you for the great simulator!

I'm currently trying to collect human-involved simulation data using OmniGibson.  
First, I opened the Omniverse UI with this python code:
```
import torch as th

import omnigibson as og
from omnigibson.robots import REGISTERED_ROBOTS

og.log.info(f"Starting")
# Create empty scene with no robots in it initially
cfg = {
    "scene": {
        "type": "Scene",
    }
}

env = og.Environment(configs=cfg)

try:
   while True: og.sim.render()
except KeyboardInterrupt:
   pass
```

Then, I tried loading human characters using the `omni.anim.people` extension, as referenced in [Isaac Sim docs](https://docs.isaacsim.omniverse.nvidia.com/4.0.0/features/warehouse_logistics/ext_omni_anim_people.html).

After I click "load characters", the human is spawned but does not move and the log shows some errors... Could you help me on fixing these errors?
 

[image]

The full log looks like this:
```
[INFO] [omnigibson.simulator] ---------- Welcome to OmniGibson! ----------
2025-03-29 22:24:58 [9,739ms] [Warning] [omni.syntheticdata.plugin] SdRenderVarPtr missing valid input renderVar LdrColorSDhost
[INFO] [omnigibson.simulator] Imported scene 0.
[18.124s] [ext: omni.kit.registry.nucleus-0.0.0] startup
[24.921s] [ext: omni.anim.behavior.schema-106.0.1] startup
[24.934s] [ext: omni.kit.widget.imageview-1.0.3] startup
[24.935s] [ext: omni.kit.window.quicksearch-2.4.4] startup
[24.939s] [ext: omni.kit.browser.sample-1.4.7] startup
[25.004s] [ext: omni.kit.quicklayout-1.0.7] startup
[25.060s] [ext: omni.graph.io-1.9.1] startup
2025-03-29 22:25:14 [25,051ms] [Warning] [omni.graph.io._impl.extension] omni.graph.io is deprecated. Nodes in this extension have been moved to omni.graph.nodes. Update any dependencies accordingly.
[25.064s] [ext: omni.anim.retarget.core-106.0.1] startup
[25.078s] [ext: omni.anim.asset-106.0.3] startup
[25.089s] [ext: omni.curve.creator-105.0.4] startup
[25.093s] [ext: omni.anim.skelJoint-106.0.1] startup
Warning: Possible version incompatibility. Attempting to load omni::fabric::IPath with version v0.2 against v0.1.
[25.109s] [ext: omni.anim.graph.core-106.0.6] startup
Warning: Possible version incompatibility. Attempting to load omni::fabric::IPath with version v0.2 against v0.1.
2025-03-29 22:25:14 [25,122ms] [Warning] [omni.graph.core._impl._registration.register_python_ogn] Python import process in omni.anim.graph.core failed - 'NoneType' object is not subscriptable. Aborting Python node registration
[25.132s] [ext: omni.curve.manipulator-105.2.6] startup
[25.211s] [ext: omni.anim.retarget.ui-106.0.1] startup
[25.275s] [ext: omni.anim.navigation.core-106.0.2] startup
[25.290s] [ext: omni.genproc.core-105.1.9] startup
[25.314s] [ext: omni.anim.retarget.bundle-106.0.1] startup
[25.317s] [ext: omni.anim.navigation.ui-106.0.2] startup
[25.431s] [ext: omni.anim.graph.ui-106.0.1] startup
[25.553s] [ext: omni.kit.scripting-106.0.1] startup
[25.558s] [ext: omni.anim.navigation.bundle-106.0.1] startup
[25.561s] [ext: omni.anim.graph.bundle-106.0.3] startup
[25.564s] [ext: omni.anim.people-0.4.1] startup
2025-03-29 22:25:48 [59,545ms] [Warning] [omni.anim.skelJoint.plugin] Obsolete SkelJoint persistent data existed in this USD stage for Skeleton /World/Characters/Biped_Setup/biped_demo_meters/Root in layer http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.1/Isaac/People/Characters/biped_demo/biped_demo_meters.usd
```