StanfordVL/BEHAVIOR-1K #243: get_position_orientation for visionsensor returns local pose

cgokmen, 2023-08-01:
As reported in #187

cremebrule, 2023-08-08:
This is a fundamental quirk of omni -- `get_position_orientation` naively computes the "world" pose by iterating through the transforms up its prim tree. This is normally fine, but when there's joints along the way, the computation is incorrect.

There's not really a unified way to compute this, since  the prim wrapped by `VisionSensor` can exist anywhere arbitrarily on the omni stage.

cgokmen, 2023-08-25:
I'm curious about this - assuming one of the parents in the tree actually exists physically (e.g. it has a pose in physx), can we not use the physx world pose of that prim (e.g. stop iterating up the tree at that point, use the transform from physx) and then go back down the tree applying whatever transforms are left?

I think this is breaking the contract of this API and is very opaque and hard to debug.

cgokmen, 2024-01-16:
To solve both this and #338 we need to add a prim registry to map prim path to prim object. Then we can replace XFormPrim's get_position_orientation to multiply its local transform with its parent's global transform

cgokmen, 2024-01-16:
Investigate: does camera work when you move robot w/ Flatcache on? Check w/ me for details of what to check

cgokmen, 2024-01-29:
Consider using the UsdRt library to get the pose

closed 2024-02-27 by hang-yin