StanfordVL/BEHAVIOR-1K #244: undefined symbol: quickReleaseFrameworkAndTerminate

AlbertTan404, 2023-08-02:
Hi! First thanks for your work!

I installed omnigibson both with docker and from source, but I no matter in which way, when I run the test python script `python -m omnigibson.examples.scenes.scene_selector `, it always returns `python: symbol lookup error: ~/.local/share/ov/pkg/isaac_sim-2022.2.0/kit/extscore/omni.gpu_foundation/bin/deps/libcarb.glinterop.plugin.so: undefined symbol: quickReleaseFrameworkAndTerminate`.
I'm running it on Ubuntu 20.04 LST with two Nvidia A6000 GPUs, and the detailed output is below:
`
[INFO] [omnigibson] ---------- Starting OmniGibson ----------
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omnigibson', 'omnigibson.macros', 'omni', 'omni.kit.app._impl.telemetry_helpers'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the following args:  ['~/.local/share/ov/pkg/isaac_sim-2022.2.0/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '~/.local/share/ov/pkg/isaac_sim-2022.2.0/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=~/.local/share/ov/pkg/isaac_sim-2022.2.0/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=False', '--/app/fastShutdown=True', '--ext-folder', '~/.local/share/ov/pkg/isaac_sim-2022.2.0/exts', '--ext-folder', '~/.local/share/ov/pkg/isaac_sim-2022.2.0/apps', '--/physics/cudaDevice=0', '--portable']
Passing the following args to the base kit application:  []
[Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[Info] [carb] Logging to file: ~/.local/share/ov/pkg/isaac_sim-2022.2.0/kit/logs/Kit/Isaac-Sim/2022.2/kit_20230802_191817.log
2023-08-02 10:18:17 [3ms] [Warning] [omni.ext.plugin] [ext: omni.drivesim.sensors.nv.lidar] Extensions config 'extension.toml' doesn't exist '~/.local/share/ov/pkg/isaac_sim-2022.2.0/exts/omni.drivesim.sensors.nv.lidar' or '~/.local/share/ov/pkg/isaac_sim-2022.2.0/exts/omni.drivesim.sensors.nv.lidar/config'
2023-08-02 10:18:17 [3ms] [Warning] [omni.ext.plugin] [ext: omni.drivesim.sensors.nv.radar] Extensions config 'extension.toml' doesn't exist '~/.local/share/ov/pkg/isaac_sim-2022.2.0/exts/omni.drivesim.sensors.nv.radar' or '~/.local/share/ov/pkg/isaac_sim-2022.2.0/exts/omni.drivesim.sensors.nv.radar/config'
[0.314s] [ext: omni.stats-0.0.0] startup
[0.367s] [ext: omni.rtx.shadercache-1.0.0] startup
[0.376s] [ext: omni.assets.plugins-0.0.0] startup
[0.378s] [ext: omni.gpu_foundation-0.0.0] startup
2023-08-02 10:18:17 [370ms] [Warning] [carb] FrameworkImpl::setDefaultPlugin(client: omni.gpu_foundation_factory.plugin, desc : [carb::graphics::Graphics v2.11], plugin : carb.graphics-vulkan.plugin) failed. Plugin selection is locked, because the interface was previously acquired by: 
[0.393s] [ext: carb.windowing.plugins-1.0.0] startup
[0.430s] [ext: omni.kit.renderer.init-0.0.0] startup
X Error of failed request:  GLXBadFBConfig
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  0 ()
  Serial number of failed request:  228
  Current serial number in output stream:  228
python: symbol lookup error: ~/.local/share/ov/pkg/isaac_sim-2022.2.0/kit/extscore/omni.gpu_foundation/bin/deps/libcarb.glinterop.plugin.so: undefined symbol: quickReleaseFrameworkAndTerminate
`
I replaced the /home/username with ~ in the above log.

Great thanks.

AlbertTan404, 2023-08-03:
> Can you tell us your GPU driver version? Version 525 is recommended for Omniverse.

I'm running on Driver Version: 520.61.05, CUDA Version: 11.8. Is it the issue?

cgokmen, 2023-08-03:
It sounds like it could be - can you try updating?

AlbertTan404, 2023-08-03:
> It sounds like it could be - can you try updating?

I'll try. Thanks!

ChengshuLi, 2023-08-14:
@AlbertTan404 Did you resolve your issue? Feel free to re-open the ticket if you still have some trouble.

Two more questions:
- Can you run Isaac Sim directly, independent of OmniGibson? Did you also observe the same error?
```
~/.local/share/ov/pkg/isaac_sim-2022.2.0
./isaac-sim.sh
```
- Does this error cause the program to crash completley? If not, it's probably okay to ignore.

closed 2023-08-14 by ChengshuLi