StanfordVL/BEHAVIOR-1K #253: Errors running Behavior 1K tasks

dmcconachie-tri, 2023-08-08:
I'm trying to experiment with the Behavior 1K tasks and I'm running into various errors depending on the task that I attempt ranging from 

```
2023-08-08 15:29:40 [148,620ms] [Error] [omni.kit.manipulator.prim.model] Traceback (most recent call last):
  File "/isaac-sim/kit/extscore/omni.kit.manipulator.prim/omni/kit/manipulator/prim/model.py", line 1492, in _update_transform_from_prims_async
    pivot_prim_path = self._pivot_prim.GetPath()
AttributeError: 'NoneType' object has no attribute 'GetPath'
```

to

```
2023-08-07 22:25:37 [388,317ms] [Error] [omni.physx.plugin] PhysX error: Non GPU compatible convex mesh isn't allowed to collied with particle system fail!!
, FILE /buildAgent/work/f25a4639a4b1bdc1/source/gpunarrowphase/src/PxgNphaseImplementationContext.cpp, LINE 1635
```
and straight up segfaults.

In all instances these errors have been replicated both in install from source and docker versions.

Replication steps:

1. `sudo ./run_docker.sh <path_to_data>` (note no `-h`, though presumably not relevant)
2. Modify `/omnigibson-src/omnigibson/macros.py` to enable `USE_GPU_DYNAMICS` based on instructions/error message when it is not enabled.
3. Modify `/omnigibson-src/omnigibson/configs/fetch_behavior.yaml` to use a particular scene
4. Start the example random motion controller `python -m omnigibson.examples.environments.behavior_env_demo`

`git log -n1` from my local (non-docker) checkout
```
commit 452b28f91a953d88d73ab4ac67560652ff5ee4f9 (HEAD -> main, upstream/main, origin/main, origin/HEAD)
Merge: 4c7e04c8 8eb9a6a3
Author: Josiah Wong <84cremebrule@gmail.com>
Date:   Sun Aug 6 23:12:35 2023 -0700

    Merge pull request #250 from StanfordVL/release-version-update
    
    Release version update
```

<details>
  <summary>Example git diff from upstream/main</summary>

Note that this includes changes to `obs_modalities` and `highlight_task_relevant_objects` but I presume this change is irrelevant. I have not gone back through and tested every scene without that, but I have run some of them with similar (perhaps identical) results.

```
diff --git a/omnigibson/configs/fetch_behavior.yaml b/omnigibson/configs/fetch_behavior.yaml
index 84da35e3..bdda7f86 100644
--- a/omnigibson/configs/fetch_behavior.yaml
+++ b/omnigibson/configs/fetch_behavior.yaml
@@ -7,7 +7,7 @@ render:
 
 scene:
   type: InteractiveTraversableScene
-  scene_model: Rs_int
+  scene_model: Beechwood_0_int
   trav_map_resolution: 0.1
   trav_map_erosion: 2
   trav_map_with_objects: true
@@ -23,7 +23,7 @@ scene:
 
 robots:
   - type: Fetch
-    obs_modalities: [scan, rgb, depth]
+    obs_modalities: [rgb]
     scale: 1.0
     self_collision: false
     action_normalize: true
@@ -50,13 +50,13 @@ objects: []
 
 task:
   type: BehaviorTask
-  activity_name: prepare_sea_salt_soak
+  activity_name: make_a_salad
   activity_definition_id: 0
   activity_instance_id: 0
   predefined_problem: null
   online_object_sampling: false
   debug_object_sampling: null
-  highlight_task_relevant_objects: false
+  highlight_task_relevant_objects: true
   termination_config:
     max_steps: 500
   reward_config:
diff --git a/omnigibson/macros.py b/omnigibson/macros.py
index 6538479e..fe76b9a6 100644
--- a/omnigibson/macros.py
+++ b/omnigibson/macros.py
@@ -40,7 +40,7 @@ gm.SHOW_DISCLAIMERS = False
 
 # Whether to use omni's GPU dynamics
 # This is necessary for certain features; e.g. particles (fluids / cloth)
-gm.USE_GPU_DYNAMICS = False
+gm.USE_GPU_DYNAMICS = True
 
 # Whether to use high-fidelity rendering (this includes, e.g., isosurfaces)
 gm.ENABLE_HQ_RENDERING = False
```

</details>

Full error messages in followup posts.

dmcconachie-tri, 2023-08-08:
<details>
  <summary>scene_model: Benevolence_1_int, activity_name: clean_a_kitchen_table</summary>

```
(omnigibson) root@geralt:/omnigibson-src#  python -m omnigibson.examples.environments.behavior_env_demo
[INFO] [omnigibson] ---------- Starting OmniGibson ----------
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omnigibson', 'omnigibson.macros', 'omni', 'omni.kit.app._impl.telemetry_helpers'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the following args:  ['/isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/isaac-sim/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/isaac-sim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=False', '--/app/fastShutdown=True', '--ext-folder', '/isaac-sim/exts', '--ext-folder', '/isaac-sim/apps', '--/physics/cudaDevice=0', '--portable', '--allow-root']
Passing the following args to the base kit application:  []
[Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[Info] [carb] Logging to file: /isaac-sim/kit/logs/Kit/Isaac-Sim/2022.2/kit_20230808_150851.log
2023-08-08 15:08:51 [2ms] [Warning] [omni.ext.plugin] [ext: omni.drivesim.sensors.nv.lidar] Extensions config 'extension.toml' doesn't exist '/isaac-sim/exts/omni.drivesim.sensors.nv.lidar' or '/isaac-sim/exts/omni.drivesim.sensors.nv.lidar/config'
2023-08-08 15:08:51 [2ms] [Warning] [omni.ext.plugin] [ext: omni.drivesim.sensors.nv.radar] Extensions config 'extension.toml' doesn't exist '/isaac-sim/exts/omni.drivesim.sensors.nv.radar' or '/isaac-sim/exts/omni.drivesim.sensors.nv.radar/config'
[0.335s] [ext: omni.stats-0.0.0] startup
[0.382s] [ext: omni.rtx.shadercache-1.0.0] startup
[0.391s] [ext: omni.assets.plugins-0.0.0] startup
[0.405s] [ext: omni.gpu_foundation-0.0.0] startup
2023-08-08 15:08:51 [439ms] [Warning] [carb] FrameworkImpl::setDefaultPlugin(client: omni.gpu_foundation_factory.plugin, desc : [carb::graphics::Graphics v2.11], plugin : carb.graphics-vulkan.plugin) failed. Plugin selection is locked, because the interface was previously acquired by: 
[0.499s] [ext: carb.windowing.plugins-1.0.0] startup
2023-08-08 15:08:52 [1,103ms] [Warning] [carb.windowing-glfw.gamepad] Joystick with unknown remapping detected (will be ignored):  Microsoft Natural® Ergonomic Keyboard 4000 [030000005e040000db00000011010000]
[1.153s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 525.125.6     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090          | Yes: 0 |     | 24810   MB | 10de      | 0          |
|     |                                  |        |     |            | 2684      | 4b710849.. |
|=============================================================================================|
| OS: Linux geralt, Version: 6.2.0-26-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
| Processor: Intel(R) Xeon(R) CPU E5-2690 v4 @ 2.60GHz | Cores: Unknown | Logical: 56
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 128771 | Free Memory: 107522
| Total Page/Swap (MB): 1951 | Free Page/Swap: 1951
|---------------------------------------------------------------------------------------------|
[3.444s] [ext: omni.kit.pipapi-0.0.0] startup
[3.467s] [ext: omni.kit.pip_archive-0.0.0] startup
[3.503s] [ext: omni.kit.loop-isaac-1.0.0] startup
[3.505s] [ext: omni.kit.async_engine-0.0.0] startup
[3.510s] [ext: omni.kit.test-0.0.0] startup
[3.744s] [ext: omni.usd.config-1.0.0] startup
[3.760s] [ext: omni.usd.libs-1.0.0] startup
[4.072s] [ext: omni.isaac.core_archive-2.0.1] startup
[4.270s] [ext: omni.pip.torch-1_13_0-0.1.4] startup
[4.277s] [ext: omni.isaac.ml_archive-1.1.0] startup
[4.510s] [ext: omni.client-0.1.1] startup
[4.540s] [ext: omni.appwindow-1.0.1] startup
[4.559s] [ext: omni.kit.renderer.core-0.0.0] startup
[4.646s] [ext: omni.kit.renderer.capture-0.0.0] startup
[4.656s] [ext: omni.kit.renderer.imgui-0.0.0] startup
[4.926s] [ext: carb.audio-0.1.0] startup
[4.933s] [ext: omni.ui-2.12.23] startup
[4.966s] [ext: omni.uiaudio-1.0.0] startup
[4.972s] [ext: omni.kit.mainwindow-1.0.0] startup
[4.980s] [ext: omni.kit.uiapp-0.0.0] startup
[4.980s] [ext: omni.usd.schema.physics-1.0.0] startup
[5.071s] [ext: omni.usd.schema.omniscripting-1.0.0] startup
[5.092s] [ext: omni.usd.schema.audio-0.0.0] startup
[5.112s] [ext: omni.usd.schema.semantics-0.0.0] startup
[5.138s] [ext: omni.usd.schema.anim-0.0.0] startup
[5.275s] [ext: omni.usd.schema.omnigraph-1.0.0] startup
[... cut ...]
Exception ignored in: <function RegisteredActions.__del__ at 0x7f086c70e830>
Traceback (most recent call last):
  File "/isaac-sim/extscache/omni.kit.viewport.menubar.lighting-104.0.8/omni/kit/viewport/menubar/lighting/actions.py", line 345, in __del__
  File "/isaac-sim/extscache/omni.kit.viewport.menubar.lighting-104.0.8/omni/kit/viewport/menubar/lighting/actions.py", line 350, in destroy
TypeError: 'NoneType' object is not callable
2023-08-08 15:15:46 [415,460ms] [Warning] [carb.audio.context] 1 contexts were leaked
^C

```

</details>

dmcconachie-tri, 2023-08-08:
`scene_model: Rs_int`

`activity_name: prepare_sea_salt_soak`

<details>
  <summary>git diff</summary>

```
(omnigibson) root@geralt:/omnigibson-src# git diff
diff --git a/omnigibson/examples/environments/behavior_env_demo.py b/omnigibson/examples/environments/behavior_env_demo.py
index fffbf876..9cd379db 100644
--- a/omnigibson/examples/environments/behavior_env_demo.py
+++ b/omnigibson/examples/environments/behavior_env_demo.py
@@ -54,4 +54,4 @@ def main(random_selection=False, headless=False, short_exec=False):
 
 
 if __name__ == "__main__":
-    main()
+    main(short_exec=True)
diff --git a/omnigibson/macros.py b/omnigibson/macros.py
index 6538479e..fe76b9a6 100644
--- a/omnigibson/macros.py
+++ b/omnigibson/macros.py
@@ -40,7 +40,7 @@ gm.SHOW_DISCLAIMERS = False
 
 # Whether to use omni's GPU dynamics
 # This is necessary for certain features; e.g. particles (fluids / cloth)
-gm.USE_GPU_DYNAMICS = False
+gm.USE_GPU_DYNAMICS = True
 
 # Whether to use high-fidelity rendering (this includes, e.g., isosurfaces)
 gm.ENABLE_HQ_RENDERING = False
```
</details>

<details>
  <summary>console output</summary>

```
(omnigibson) root@geralt:/omnigibson-src# python -m omnigibson.examples.environments.behavior_env_demo
[INFO] [omnigibson] ---------- Starting OmniGibson ----------
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omnigibson', 'omnigibson.macros', 'omni', 'omni.kit.app._impl.telemetry_helpers'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the following args:  ['/isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/isaac-sim/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/isaac-sim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=False', '--/app/fastShutdown=True', '--ext-folder', '/isaac-sim/exts', '--ext-folder', '/isaac-sim/apps', '--/physics/cudaDevice=0', '--portable', '--allow-root']
Passing the following args to the base kit application:  []
[Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[Info] [carb] Logging to file: /isaac-sim/kit/logs/Kit/Isaac-Sim/2022.2/kit_20230808_162739.log
2023-08-08 16:27:39 [2ms] [Warning] [omni.ext.plugin] [ext: omni.drivesim.sensors.nv.lidar] Extensions config 'extension.toml' doesn't exist '/isaac-sim/exts/omni.drivesim.sensors.nv.lidar' or '/isaac-sim/exts/omni.drivesim.sensors.nv.lidar/config'
2023-08-08 16:27:39 [2ms] [Warning] [omni.ext.plugin] [ext: omni.drivesim.sensors.nv.radar] Extensions config 'extension.toml' doesn't exist '/isaac-sim/exts/omni.drivesim.sensors.nv.radar' or '/isaac-sim/exts/omni.drivesim.sensors.nv.radar/config'
[0.294s] [ext: omni.stats-0.0.0] startup
[0.339s] [ext: omni.rtx.shadercache-1.0.0] startup
[0.345s] [ext: omni.assets.plugins-0.0.0] startup
[0.347s] [ext: omni.gpu_foundation-0.0.0] startup
2023-08-08 16:27:39 [338ms] [Warning] [carb] FrameworkImpl::setDefaultPlugin(client: omni.gpu_foundation_factory.plugin, desc : [carb::graphics::Graphics v2.11], plugin : carb.graphics-vulkan.plugin) failed. Plugin selection is locked, because the interface was previously acquired by: 
[0.358s] [ext: carb.windowing.plugins-1.0.0] startup
2023-08-08 16:27:40 [926ms] [Warning] [carb.windowing-glfw.gamepad] Joystick with unknown remapping detected (will be ignored):  Microsoft Natural® Ergonomic Keyboard 4000 [030000005e040000db00000011010000]
[0.943s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 525.125.6     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090          | Yes: 0 |     | 24810   MB | 10de      | 0          |
|     |                                  |        |     |            | 2684      | 4b710849.. |
|=============================================================================================|
| OS: Linux geralt, Version: 6.2.0-26-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
| Processor: Intel(R) Xeon(R) CPU E5-2690 v4 @ 2.60GHz | Cores: Unknown | Logical: 56
[... 540 lines cut ...]
2023-08-08 16:30:15 [155,839ms] [Warning] [omni.physx.plugin] Non-uniform scale may result in a non matching collision representation on prim: /World/electric_switch_wseglt_5/togglebutton_0_0_link/mesh_0
2023-08-08 16:30:15 [155,839ms] [Warning] [omni.physx.plugin] Non-uniform scale may result in a non matching collision representation on prim: /World/electric_switch_wseglt_2/togglebutton_0_0_link/mesh_0
[... cut ...]
 [omni.kit.manipulator.prim.model] Traceback (most recent call last):
  File "/isaac-sim/kit/extscore/omni.kit.manipulator.prim/omni/kit/manipulator/prim/model.py", line 1492, in _update_transform_from_prims_async
    pivot_prim_path = self._pivot_prim.GetPath()
AttributeError: 'NoneType' object has no attribute 'GetPath'

[INFO] [omnigibson] ---------- Shutting Down OmniGibson ----------
[157.461s] Simulation App Shutting Down
```
</details>

Oddities:
- Large salt particles
- Salt exploding out of the shaker
- Couch in the floor

dmcconachie-tri, 2023-08-08:
[Screencast from 08-08-2023 12:28:10 PM.webm](https://github.com/StanfordVL/OmniGibson/assets/71032891/ef27fb44-dea2-417e-98c4-09e2cc4ae4b6)

cremebrule, 2023-08-08:
Hi @dmcconachie-tri ,

Huge thanks for taking the time to compile these bug reports. Our team is taking a look into it; we should hopefully have fixes ready by tomorrow!

cremebrule, 2023-09-05:
Hi @dmcconachie-tri ,

Apologies for the delay. We have a new release out ([0.2.1](https://github.com/StanfordVL/OmniGibson/releases/tag/v0.2.1)) that fixes some of your issues. Concretely:

----

`scene_model: Benevolence_1_int`

`activity_name: clean_a_kitchen_table`

I was unable to deterministically reproduce this error on my machine. It occurred the first time I ran but did not show up during subsequent runs. This is a bit concerning, as it seems to suggest potentially an NVIDIA / Omniverse caching issue of some sort. We are hoping it can be reproduced potentially on another setup (@cgokmen )

----

`scene_model: Beechwood_0_int`

`activity_name: make_a_salad`

I was able to reproduce the segfaults deterministically. We had similar issues with IsaacSim v2022.2.1 (stochastically) but had never seen it occur with 2022.2.0. We're hoping that the impending NV release v2023.1.0 will resolve this issue.

----

`scene_model: Rs_int`

`activity_name: prepare_sea_salt_soak`

This (as well as other problematic tasks with similar object-through-floor) has been resampled. Moreover, salt should no longer explode out of the container.

Regarding salt particle sizes, it is a known limitation of the NVIDIA Omniverse particle systems. We could reduce the diameter of the particles, but physics stability is likely to decrease, especially at the default physics timestep size (1/60). In general, we've tried to limit our particles from being less than 1-2cm in diameter for this reason, at the expense of a bit less realism. Currently there's no explicit API that allows you to modify the particle size at runtime, but you can always go to the hyperparameter specification (in `<DATASET>/systems/<system_name>/metadata.json`) and directly modify the size there.

----

As an aside, there's also been other problematic tasks (not mentioned explicitly in this thread) and issues that have been fixed as well. Apologies for the issues; again, huge thanks for tagging these bugs and being patient with the development. We're planning on having another mini-release soon once NVIDIA releases their new IsaacSim version -- hopefully within the next 3 weeks or so!

dajiebi, 2023-11-01:
May I ask where I can run the Behavior 1K tasks？

lyh1028, 2023-11-26:
It seems these problems haven't been resolved... I install Omnigibson with docker on linux without GUI as the link [OmniGibson Documentation](https://behavior.stanford.edu/omnigibson/getting_started/installation.html#requirements). However when I run `python -m omnigibson.examples.environments.behavior_env_demo` (the only change is to enable USE_GPU_DYNAMICS), I also meet the error: 
`2023-11-26 11:58:24 [1,146,503ms] [Error] [omni.kit.manipulator.prim.model] Traceback (most recent call last):
  File "/isaac-sim/kit/extscore/omni.kit.manipulator.prim/omni/kit/manipulator/prim/model.py", line 1492, in _update_transform_from_prims_async
    pivot_prim_path = self._pivot_prim.GetPath()
AttributeError: 'NoneType' object has no attribute 'GetPath'`

cgokmen, 2024-01-11:
I believe all of the above issues are fixed in our current development branch - closing this issue, please keep an eye out for our upcoming release.

closed 2024-01-11 by cgokmen