StanfordVL/BEHAVIOR-1K #254: Failed to run tutorial example.

WangYixuan12, 2023-08-08:
Hi! Thank you for your hard work! When I followed the installation instruction [here](https://stanfordvl.github.io/OmniGibson/getting_started/installation.html#setup) and run `python -m omnigibson.examples.scenes.scene_selector`, I encountered the error
```
FileNotFoundError: [Errno 2] No such file or directory: '/home/yixuan/OmniGibson/omnigibson/data/og_dataset/objects/bottom_cabinet/bamfsz/usd/bamfsz.encrypted.usd'
```
I checked the folder. It seems `bottom_cabinet` is missing. But I downloaded og_dataset by following the instruction. Could you please help with this? Thank you in advance for your help!

cremebrule, 2023-08-08:
Hi @WangYixuan12 ,

Thanks for raising this issue! Did you install from docker or from source?

cremebrule, 2023-08-08:
I just tried the docker install and I can't reproduce your error -- did you specify a custom DATA_DIR?

WangYixuan12, 2023-08-09:
Thank you @cremebrule ! I installed from source. Any information I could provide to help you reproduce the error?

cremebrule, 2023-08-09:
I see -- have you verified that the dataset has been downloaded at all? Or did the dataset fail to download altogether?

WangYixuan12, 2023-08-09:
I just went through the installation process again. This time the error did not appear. The reason might be that I experienced download interruption for the first time, and resuming downloading did not fully download the whole dataset. Sorry for the confusion!

closed 2023-08-09 by WangYixuan12