StanfordVL/BEHAVIOR-1K #264: Enabling livestream in v0.2.0

sai-prasanna, 2023-08-25:
I was able to get livestream working in headless mode using 

```
from omni.isaac.core.utils.extensions import enable_extension
enable_extension("omni.kit.livestream.native")
```

But now, with v0.2.0, I get the following error and the isaac-sim process dies the moment I click connect in the streaming client.

```
python: symbol lookup error: /isaac-sim/kit/extscore/omni.kit.streamsdk.plugins/bin/libssl.so.1.1: undefined symbol: EVP_idea_cbc, version OPENSSL_1_1_0 
```

Any help here is much appreciated. I can't replicate it working in v0.1.0 to compare because the docker image for v0.1.0 has /omnigibson-src with different source code, I reset it to v0.1.0 manually and then the dataset of v0.2.0 is not compatible with older image, but there is no way to download the older dataset as the url is the same.

sai-prasanna, 2023-08-25:
I managed to fix it by installing openssl 1.1 in the conda environment. I think isaac assumes that openssl version is 1.1  and somewhere in the current setup, a newer vesion is being installed.

cgokmen, 2023-08-25:
This is a really interesting use case! We'll fix the openssl situation - hopefully we can also document the livestream use too later & we will try to implement a dataset URL per version going forward to avoid this issue. Thanks!

sai-prasanna, 2023-08-26:
Yep, this is really useful when you wish to access your server remotely but want to use isaac's GUI.

cgokmen, 2024-01-11:
There is no issue with OpenSSL in the latest og-develop copy of OmniGibson, this feature works reliably now. Closing.

closed 2024-01-11 by cgokmen