StanfordVL/BEHAVIOR-1K #266: Change resolution of Camera Sensor

KayJersch, 2023-08-27:
By default the resolution of Fetchs Camera Sensor is 128x128. That is too small for what I'm trying to do. Is there any way to increase the resolution?

cgokmen, 2023-08-30:
There are image_height and image_width on the VisionSensor class. You can access the visionsensor using the robot.sensors dictionary.

cgokmen, 2023-08-30:
@cremebrule do we have a good way of configuring the sensor resolution from the config file currently?

KayJersch, 2023-09-01:
> There are image_height and image_width fields on the VisionSensor that you can set. You can access the visionsensor using the robot.sensors dictionary.

using
`env.robots[0].sensors['robot0:eyes_Camera_sensor'].image_height = 512
env.robots[0].sensors['robot0:eyes_Camera_sensor'].image_width =  512`
causes the following error:
`[ERROR] [omnigibson.envs.env_base] OBSERVATION SPACE:
[ERROR] [omnigibson.envs.env_base] ('obs_space', 'robot0::robot0:eyes_Camera_sensor_rgb', dtype('uint8'), (128, 128, 4))
[ERROR] [omnigibson.envs.env_base] ('obs_space', 'task::low_dim', dtype('float64'), (0,))
[ERROR] [omnigibson.envs.env_base] ACTUAL OBSERVATIONS:
[ERROR] [omnigibson.envs.env_base] ('obs', 'robot0::robot0:eyes_Camera_sensor_rgb', dtype('uint8'), (512, 512, 4))
[ERROR] [omnigibson.envs.env_base] ('obs', 'task::low_dim', dtype('float64'), (0,))
Traceback (most recent call last):
  File "main.py", line 72, in <module>
    obs = env.reset()['robot0']['robot0:eyes_Camera_sensor_rgb']
  File "/home/MYNAME/OmniGibson/omnigibson/envs/env_base.py", line 452, in reset
    raise ValueError("Observation space does not match returned observations!")
ValueError: Observation space does not match returned observations!
[INFO] [omnigibson] ---------- Shutting Down OmniGibson ----------
[25.031s] Simulation App Shutting Down`
How do I change the shape of the observation space?

cgokmen, 2023-09-01:
We'll fix this today and let you know and you can check out the development code.

KayJersch, 2023-09-12:
I tried reinstalling OmniGibson (deleting the environment, deleting the the cloned omnigibson repository and downloading it again) and now `setup.sh` gives me this error:
`./scripts/setup.sh: line 30: /home/MYNAME/.local/share/ov/pkg/isaac_sim-2022.2.0/python: No such file or directory`

cgokmen, 2023-09-12:
Looks like a bug with one of our new features. Hotfix for you, replace this: https://github.com/StanfordVL/OmniGibson/blob/aa8b8701e6abc3f2310a883a0ce2b635405347c7/scripts/setup.sh#L30-L31

with `ISAAC_PYTHON_VERSION=3.7`

Separately, to help us fully solve the issue, can you take a look inside `/home/MYNAME/.local/share/ov/pkg/isaac_sim-2022.2.0/` and let us know if there is any kind of a python executable anywhere in there?

KayJersch, 2023-09-12:
That worked! Thanks! (also there is a `python.sh` file in `/home/MYNAME/.local/share/ov/pkg/isaac_sim-2022.2.0/` is that what you mean?)

cgokmen, 2023-09-12:
I see, yes that's what I was looking for. We'll fix it, thanks!

KayJersch, 2023-09-12:
I still have the same error: `ValueError: Observation space does not match returned observations!`
using the same code:
```
env.robots[0].sensors['robot0:eyes_Camera_sensor'].image_height = 512
env.robots[0].sensors['robot0:eyes_Camera_sensor'].image_width =  512
```
is there anything else I need to add?

cgokmen, 2023-09-12:
Oh, sorry I forgot to tell you what to do. Remove those lines and set the resolution in your config file this way: https://github.com/StanfordVL/OmniGibson/blob/806ba6159f688b16eaa96dd865d80413691928fc/omnigibson/configs/fetch_behavior.yaml#L36-L40

KayJersch, 2023-09-13:
That did not work

cgokmen, 2023-09-13:
Please give us some info to work with here - did it crash? Or does it work but the image still the wrong resolution?

KayJersch, 2023-09-14:
It runs just fine but the image is still 128x128

cremebrule, 2023-09-15:
Hi @KayJersch ,

Could you provide us with the .yaml config you're using? I will try to reproduce the problem on our end. Thanks!

KayJersch, 2023-09-15:
@cremebrule  sure!
[fetch_behavior.zip](https://github.com/StanfordVL/OmniGibson/files/12629187/fetch_behavior.zip)

cremebrule, 2023-09-22:
Hi @KayJersch ,

Ah! It's because, by default, loading a BEHAVIOR Task (or, more generally, any pre-cached scene) uses the robot already included in that file.

To force your agent to be used instead, you can set the [`scene` config's `include_robots=False`](https://github.com/StanfordVL/OmniGibson/blob/og-develop/omnigibson/configs/fetch_behavior.yaml#L22).

Can you try that and let us know if it works for you?

KayJersch, 2023-09-24:
I tried it, but the resolution stays the same

ChengshuLi, 2024-09-23:
Feel free to re-open the issue if the problem persists, thanks!

closed 2024-09-23 by ChengshuLi