StanfordVL/BEHAVIOR-1K #267: question about the trajectory calculated by IKSolver

FinnJob, 2023-08-28:
Thank you for your excellent work! I have some questions regarding the use of the Fetch robotic arm in this simulation environment:

- I've been using an IKSolver to compute joint values for a target pose. However, the trajectory of the robotic arm seems a bit odd. For a small movement along a particular coordinate, there seems to be a significant motion of the arm. How can I prevent this from happening?
- How should I plan the motion to ensure the path is collision-free?

Would you be able to provide some guidance on these matters? Thank you.

cgokmen, 2023-08-30:
@sujaygarlanka can you help?

StarsTesla, 2023-08-31:
Same here，Is there a way to control or select a non-collision solution? And another problem is that the gripper of fetch seems to be not equal power when I try to grip the cube, it would always have some offset from the central point.

mj-hwang, 2023-09-03:
Hello,

I can provide some insights and guidelines regarding above questions:

* We use [IK Solver](https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.lula/docs/index.html#lula.compute_ik_ccd) from Omniverse's motion & kinematics library Lula, as shown in our source code for inverse kinematics(). This uses cyclic coordinate descent algorithm to execute the inverse kinematics. To answer your question, as shown in the [docstring](https://github.com/StanfordVL/OmniGibson/blob/main/omnigibson/utils/control_utils.py#L28C13-L28C13), we can do the followings to avoid the jerky motion with IK.
    - We can specify initial_joint_pos to be the previous joint position to minimize jerky motions.
    - Try to plan in a higher frequency so that the next goal end effector position & quaternion is close to the current one. I found that quaternion matters a lot, so it is recommended to set the target quaternion, and to set it close to the previous one if you can. 
    - You can adjust tolerance, weight, and max iteration parameters depending on the goal.
* For computing a collision-free motion plan, you can use motion planning libraries (e.g. [OMPL](https://ompl.kavrakilab.org/)) to plan a motion that avoids external collisions and self-collisions. One way to do this is shown in one of our development branches, in which we are implementing a set of action primitives. We generate plans in the joint space, and we check if a certain joint configuration is valid by using forward kinematics. This is done by checking whether mesh overlaps exist or not, except for an object held in a gripper or some pre-defined self-collision pairs of robot links. More details can be found in here: [link1](https://github.com/StanfordVL/OmniGibson/blob/bc11b7e9736a95319451650c3d73c81a1a91edd7/omnigibson/utils/motion_planning_utils.py#L225), [link2](https://github.com/StanfordVL/OmniGibson/blob/bc11b7e9736a95319451650c3d73c81a1a91edd7/omnigibson/utils/motion_planning_utils.py#L339). Note that this branch is still under development.
* In order to compute collision-aware motion plans, we shall use JointSpaceController and use forward kinematics to check for collisions. If we use IK, there might be a number of different joint configurations that fulfill a specific EEF pose, among which some are valid, while some are not.

Please let us know if you have any question, and feel free to make PRs if you extend or improve some of above features (IK, collision-free planning). Thanks!

Best,
Minjune

StarsTesla, 2023-09-05:
@mj-hwang Thanks for your attention, we will have a try this week of the collision free motion planner.

FinnJob, 2023-09-11:
@mj-hwang Thank you, it helps a lot.

closed 2023-10-16 by wensi-ai