StanfordVL/BEHAVIOR-1K #271: request for checkpoint

Draculair, 2023-09-08:
Hi, Do you have the pre-trained checkpoint of the baseline from the experiments in your paper in the repository? Thanks.

cgokmen, 2023-09-13:
Unlikely (since a lot of code has been changed since then), but @j96w for visibility

Draculair, 2023-10-04:
@j96w Can I trouble you for the checkpoints in your paper? Thanks

j96w, 2023-10-05:
Hey @Draculair, sorry for the late response. Due to the large changes over the code base, the trained checkpoints (trained in Sep. 2021) are no longer compatible with the current infra (released in Seq 2023).

Draculair, 2023-10-05:
What about the latest version, is it availale? Or can you tell me how to train the basic skills in the benchmark? Thank you!

j96w, 2023-10-12:
Hey @Draculair, we didn't train skills in our previous submission, the model is trained with RL to generate high-level skill selection instead of actual robot motor commands. You can check out our study paper for more information about that. About the latest version, we do have several on-going projects but most of them are not following the high-level skill training fashion anymore due to the rapid development of new decision-making methods. We will let you know when we are ready to release more.

Draculair, 2023-10-12:
Hey @j96w, I'd like to ask how you drive the robot to perform these decomposed actions (skills) when you have done the task decomposition (decision making)? Thanks

cgokmen, 2023-10-12:
You can take a look at our ongoing effort to produce action primitives. This is in development currently but it'll give you an idea of how you can implement these things (e.g. driving the robot around). https://github.com/StanfordVL/OmniGibson/blob/action-primitives/omnigibson/action_primitives/starter_semantic_action_primitives.py

closed 2023-10-12 by cgokmen