StanfordVL/BEHAVIOR-1K #283: No file named 'setup_conda_env.sh'

Draculair, 2023-09-27:
[image]
I'm following the instuctions in https://behavior.stanford.edu/omnigibson/getting_started/installation.html#setup to install the omnigibson and I found that the `setup_conda_env.sh` is missed, who can help? thanks

wensi-ai, 2023-09-28:
Hi @Draculair, that documentation is a bit outdated, please use [this updated link](https://stanfordvl.github.io/OmniGibson/getting_started/installation.html#setup) instead.

Draculair, 2023-10-03:
> Hi @Draculair, that documentation is a bit outdated, please use [this updated link](https://stanfordvl.github.io/OmniGibson/getting_started/installation.html#setup) instead.

I've encountered a new problem. When I run `./scripts/setup.sh`, This error occured:
```
(base) draculair@draculair-ubuntu20:~/OmniGibson$ ./scripts/setup.sh 
We found Isaac Sim installed at /home/draculair/.local/share/ov/pkg/isaac_sim-2022.2.0. OmniGibson will use it by default.
If you want to use a different one, please type in the path containing isaac-sim.sh here (press enter to skip) >>> 

Using Isaac Sim at /home/draculair/.local/share/ov/pkg/isaac_sim-2022.2.0

The new conda environment will be named omnigibson by default.
If you want to use a different name, please type in here (press enter to skip) >>> 

Using omnigibson as the conda environment name

./scripts/setup.sh: 行 30: /home/draculair/.local/share/ov/pkg/isaac_sim-2022.2.0/python: No such file or directory.
```
Can you help? Thank you

cgokmen, 2023-10-03:
Hi, can you checkout the og-develop branch and try again? This bug was fixed a couple weeks ago but we have not had a chance to release it.

Also we released the docs again on the original link you used. Thanks for the heads up.

yuanfei-Wang, 2023-10-09:
Hello, I followed the [updated instructions](https://stanfordvl.github.io/OmniGibson/getting_started/installation.html), and I have checkout the branch to og-develop. But I still have a problem when downloading OmniGibson dataset. First, the process failed due to 'scipy' module missing. After I pip installed the module, the process again failed and suggested that the 'omni' module is missing.

Draculair, 2023-10-09:
I don't know why but you have to run it in the terminal but not in vscode.

yuanfei-Wang, 2023-10-09:
> I don't know why but you have to run it in the terminal but not in vscode.

Actually, I did run the command in the terminal on a Linux machine. Did you succeed by checkout the og-develop branch?

Draculair, 2023-10-09:
yeah, I've checkout to og-develop.

cgokmen, 2023-10-09:
You do have Isaac Sim installed and you can launch it from the Omniverse Launcher, correct?

yuanfei-Wang, 2023-10-09:
> You do have Isaac Sim installed and you can launch it from the Omniverse Launcher, correct?

Yes I have installed Isaac Sim and I can launch it.

cgokmen, 2023-10-09:
When you call the omnigibson setup script it finds the right version of Isaac Sim? Also when you go into the Python interpreter in the omnigibson env and you do `import omni` what happens?

yuanfei-Wang, 2023-10-09:
> When you call the omnigibson setup script it finds the right version of Isaac Sim? Also when you go into the Python interpreter in the omnigibson env and you do `import omni` what happens?

It seems that the setup script finds the right version of Isaac Sim.
[image]
After the omnigibson environment is created, I call the conda activate command and the following error occurs.
[image]
If I ignore this error and continue to run the command to download the dataset, the process collapses due to 'scipy' module missing:
[image]
After I pip install scipy and rerun the download command, it says the omni module is missing:
[image]
If I go into the python interpreter in omnigibson env and import omni, the same error occurs:
[image]

wensi-ai, 2023-10-09:
Hi @yuanfei-Wang, are you using bash or zsh? Could you show us your `~/.local/share/ov/pkgs/isaac_sim-2022.2.0/setup_conda_env.sh`?

cgokmen, 2023-10-09:
Yes this stuff should be executed using bash. You can go into a bash session (bash --login) and do it there.

yuanfei-Wang, 2023-10-10:
Yes, I was using zsh. After changing into a bash session, the above errors disappeared. Thanks a lot!

closed 2023-10-10 by cgokmen