StanfordVL/BEHAVIOR-1K #294: When I finished downloading, I had some problems running the code

Hollow-j, 2023-10-19:
After I had downloaded the latest download instructions, I tried to run the first command（python -m omnigibson.examples.scenes.scene_selector ） in “explore omnigibson” and got the following errors
[image]
[image]
But eventually, I got to this interface（）
[image]
When I've made all the choices,
[image]
[image]
and then It finally closed
[image]

Hollow-j, 2023-10-19:
can anyone help？I've been troubled for a long time.

cgokmen, 2023-10-19:
Hi,

Everything here is running as expected - those red errors are from some part of Isaac Sim we don't use, and apart from that this script executes to completion (I know nothing too interesting happens)!

If you want to drive a robot around a bit as a demo, run `python -m omnigibson.examples.robots.robot_control_example`

Thanks!

Hollow-j, 2023-10-20:
I see. Thank you very much for your reply！