StanfordVL/BEHAVIOR-1K #301: Incorrect sensor return

StarsTesla, 2023-10-24:
Hello, I am using robot sensors to detect surrounding objects, boxes=self. robot. get_ OBS() ['Fetch: eyes_Camera_sensor_bbox_2d_right '], but I found that the returned Box and object name do not match correctly. This has caused problems with my application because I am unable to correctly recognize and process detected objects.

cgokmen, 2023-10-24:
I will be investigating this this week - what version of Isaac Sim are you on?

StarsTesla, 2023-10-25:
2022.0

StarsTesla, 2023-10-25:
> I will be investigating this this week - what version of Isaac Sim are you on?

2022.0

cgokmen, 2024-01-16:
Hi, can you give us a script that we can try out to reproduce this issue? We were unable to reproduce this. Sorry for the late response.

cgokmen, 2024-01-16:
(for OG team: this is related to #458 )

ChengshuLi, 2024-09-23:
Should be fixed now.

closed 2024-09-23 by ChengshuLi