StanfordVL/BEHAVIOR-1K #302: isaac_sim-2022.2.0/python: No such file or directory

0nhc, 2023-10-24:
My system version is Ubuntu 20.04.

I was following the installation guide: [https://behavior.stanford.edu/omnigibson/getting_started/installation.html#setup](https://behavior.stanford.edu/omnigibson/getting_started/installation.html#setup).

However after I executed this command in my bash terminal:
```sh
./scripts/setup.sh
```
The terminal outputs this error:
```sh
We found Isaac Sim installed at /home/hzx/.local/share/ov/pkg/isaac_sim-2022.2.0. OmniGibson will use it by default.
If you want to use a different one, please type in the path containing isaac-sim.sh here (press enter to skip) >>> 

Using Isaac Sim at /home/hzx/.local/share/ov/pkg/isaac_sim-2022.2.0

The new conda environment will be named omnigibson by default.
If you want to use a different name, please type in here (press enter to skip) >>> 

Using omnigibson as the conda environment name

./scripts/setup.sh: line 30: /home/hzx/.local/share/ov/pkg/isaac_sim-2022.2.0/python: No such file or directory
```

I thought it was because of broken packages of my Isaac Sim 2022.2.0 that led to this python path error. But even after I reinstalled Isaac Sim 2022.2.0, the error still exists. So I modified line 30 of [./scripts/setup.sh](https://github.com/StanfordVL/OmniGibson/blob/main/scripts/setup.sh) and it finally works.
```sh
# ISAAC_PYTHON_VERSION=$(${ISAAC_SIM_PATH}/python -c "import platform; print(platform.python_version())")
ISAAC_PYTHON_VERSION=$(${ISAAC_SIM_PATH}/kit/python/bin/python3 -c "import platform; print(platform.python_version())")
```

0nhc, 2023-10-24:
pr #303

wensi-ai, 2023-10-24:
Hi @0nhc , We have fixed this issue on `og-develop` but haven't merged to `main` yet, so please try install on `og-develop` instead. Sorry for the inconvenience.

closed 2023-10-24 by wensi-ai