StanfordVL/BEHAVIOR-1K #321: Husky controller

lhy0807, 2023-11-13:
Thanks for this amazing project! I have a question regarding the controller for Husky, which is currently implemented using `JointController` class. However, seems like [it should be](https://github.com/husky/husky/blob/100563b7e4fe0867b8b8daef761656b7f8b89950/husky_control/config/control.yaml#L6C3-L6C52) `DifferentialDrive`.

I wonder is there any consideration behind using `JointController`? Should I attempt to implement `DifferentialDrive` for Husky? Thanks a lot!

cgokmen, 2023-12-13:
In OmniGibson, you can always configure robots to use different controllers just by changing the controller config in your config file. But the main reason why the Husky/DD configuration doesn't currently work is because the DifferentialDriveController expects two wheels for differential drive (e.g. like Fetch, Turtlebot, etc) whereas the Husky has 4 wheels. You can implement this feature if you'd like by editing the DifferentialDriveController class to output n controls instead of always 2. Feel free to create a PR into the og-develop branch if you end up doing this.

cgokmen, 2023-12-13:
Leaving this open as a feature request

ChengshuLi, 2024-09-23:
Currently `DifferentialDrive` only supports two-wheeled robots. If you figure out a way to support Husky, which is four-wheeled, feel free to open a PR and we're happy to incorporate it, thanks!!

closed 2024-09-23 by ChengshuLi