StanfordVL/BEHAVIOR-1K #332: Grasping and base movement of robots

chenjiayi-zxx, 2023-12-06:
Hello, I've encountered some issues while using the Fetch robot. I've been using methods like turn_left to move my robot, but when the robot's manipulator arm grasps objects, the objects often slip out of the arm's control. This problem prevents the robot from moving objects stably. I'm wondering if there are any solutions or best practices to address this issue. Thank you very much for your assistance.

cgokmen, 2023-12-13:
Hi, can you provide a script to reproduce the issue? Thanks!

ChengshuLi, 2024-07-16:
Closing the issue for now. Feel free to re-open if you have further questions.

In general, if the grasp is not firm enough during navigation, maybe you can consider using `"assitive"` or "sticky"` grasping_mode (see `manipulation_robot.py` for more details).

closed 2024-07-16 by ChengshuLi