StanfordVL/BEHAVIOR-1K #354: Bug related to setting joint target

wensi-ai, 2023-12-23:
Issue related to this [pull request](https://github.com/StanfordVL/OmniGibson/pull/353). Currently we need to set the joint effort targets all at once bc if we set the joint target sequentially the previous changes will be overwritten.

How to reproduce:

change line 461 of `controllable_object.py` from

`self.set_joint_efforts(efforts=np.array(eff_vec), indices=np.array(eff_idxs), normalized=normalized)`

to

`for eff, idx in zip(eff_vec, eff_idxs): self.set_joint_efforts(efforts=eff, indices=idx, normalized=normalized)`

and run `robot_control_example.py`, choose FrankaPanda and the OSC controller will not work

cgokmen, 2024-01-16:
Currently fixed on our end by batching those writes but need to figure out root cause

cgokmen, 2024-01-17:
This actually affects position and velocity as well. High priority

cgokmen, 2024-01-17:
Setting the joint position of any subset of joints on a robot causes all of the robot's joint position targets to move to their current positions too, making position control have no effect on the robot. Same issue applies to velocities and velocity targets. It's a bug in the omni.isaac.core.ArticulationView class, causes pretty much all control to break under AG. Elevated this to P0, fixing it now

cgokmen, 2024-01-17:
Root cause explained in https://forums.developer.nvidia.com/t/articulationview-bugs-in-joint-position-velocity-effort-setting/279186

closed 2024-01-17 by cgokmen