StanfordVL/BEHAVIOR-1K #559: GUI does not show up when running example from docker

tony88k, 2024-01-21:
I followed [the installation instructions](https://behavior.stanford.edu/omnigibson/getting_started/installation.html#setup),
and set up the docker container. When running the example
```
python -m omnigibson.examples.scenes.scene_selector
```
From the doc's Troubleshooting section, looks like the Omniverse Launcher app should pop up.  However, this is not happening for me. Below is a dump of my terminal output.
```
[INFO] [omnigibson] ---------- Starting OmniGibson ----------
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omnigibson', 'omnigibson.macros', 'omni', 'omni.kit.app._impl.telemetry_helpers'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the following args:  ['/isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/isaac-sim/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/isaac-sim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=False', '--/app/fastShutdown=True', '--ext-folder', '/isaac-sim/exts', '--ext-folder', '/isaac-sim/apps', '--/physics/cudaDevice=0', '--portable', '--no-window', '--allow-root']
Passing the following args to the base kit application:  []
[Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[Info] [carb] Logging to file: /isaac-sim/kit/logs/Kit/Isaac-Sim/2022.2/kit_20240121_131643.log
2024-01-21 13:16:43 [1ms] [Warning] [omni.ext.plugin] [ext: omni.drivesim.sensors.nv.lidar] Extensions config 'extension.toml' doesn't exist '/isaac-sim/exts/omni.drivesim.sensors.nv.lidar' or '/isaac-sim/exts/omni.drivesim.sensors.nv.lidar/config'
2024-01-21 13:16:43 [2ms] [Warning] [omni.ext.plugin] [ext: omni.drivesim.sensors.nv.radar] Extensions config 'extension.toml' doesn't exist '/isaac-sim/exts/omni.drivesim.sensors.nv.radar' or '/isaac-sim/exts/omni.drivesim.sensors.nv.radar/config'
[0.133s] [ext: omni.stats-0.0.0] startup
[0.164s] [ext: omni.rtx.shadercache-1.0.0] startup
[0.170s] [ext: omni.assets.plugins-0.0.0] startup
[0.172s] [ext: omni.gpu_foundation-0.0.0] startup
2024-01-21 13:16:43 [164ms] [Warning] [carb] FrameworkImpl::setDefaultPlugin(client: omni.gpu_foundation_factory.plugin, desc : [carb::graphics::Graphics v2.11], plugin : carb.graphics-vulkan.plugin) failed. Plugin selection is locked, because the interface was previously acquired by: 
[0.182s] [ext: carb.windowing.plugins-1.0.0] startup
2024-01-21 13:16:43 [169ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-01-21 13:16:43 [170ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.3]) (impl: carb.windowing-glfw.plugin)
[0.183s] [ext: omni.kit.renderer.init-0.0.0] startup
2024-01-21 13:16:43 [236ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-01-21 13:16:43 [236ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.3]) (impl: carb.windowing-glfw.plugin)
2024-01-21 13:16:43 [236ms] [Error] [carb.glinterop.plugin] GLInteropContext::init: carb::windowing is not available
2024-01-21 13:16:43 [236ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-RTX GPU: NVIDIA GeForce GTX 1050 Ti
2024-01-21 13:16:43 [236ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-RTX GPU: NVIDIA GeForce GTX 1050 Ti

|---------------------------------------------------------------------------------------------|
| Driver Version: 470.223.2     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 3080          | Yes: 0 |     | 10486   MB | 10de      | 0          |
|     |                                  |        |     |            | 2216      | e687e32a.. |
|---------------------------------------------------------------------------------------------|
| 1   | NVIDIA GeForce GTX 1050 Ti       |        |     | 4342    MB | 10de      | 0          |
|     |                                  |        |     |            | 1c82      | 8eddf50b.. |
|=============================================================================================|
| OS: Linux alicander, Version: 5.4.0-169-generic
| Processor: AMD Ryzen Threadripper 1900X 8-Core Processor   | Cores: Unknown | Logical: 16
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 31952 | Free Memory: 27894
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2047
|---------------------------------------------------------------------------------------------|
[INFO] [omnigibson] ---------- Starting OmniGibson ----------
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omnigibson', 'omnigibson.macros', 'omni', 'omni.kit.app._impl.telemetry_helpers'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the following args:  ['/isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/isaac-sim/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/isaac-sim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=False', '--/app/fastShutdown=True', '--ext-folder', '/isaac-sim/exts', '--ext-folder', '/isaac-sim/apps', '--/physics/cudaDevice=0', '--portable', '--no-window', '--allow-root']
Passing the following args to the base kit application:  []
[Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[Info] [carb] Logging to file: /isaac-sim/kit/logs/Kit/Isaac-Sim/2022.2/kit_20240121_131643.log
2024-01-21 13:16:43 [1ms] [Warning] [omni.ext.plugin] [ext: omni.drivesim.sensors.nv.lidar] Extensions config 'extension.toml' doesn't exist '/isaac-sim/exts/omni.drivesim.sensors.nv.lidar' or '/isaac-sim/exts/omni.drivesim.sensors.nv.lidar/config'
2024-01-21 13:16:43 [2ms] [Warning] [omni.ext.plugin] [ext: omni.drivesim.sensors.nv.radar] Extensions config 'extension.toml' doesn't exist '/isaac-sim/exts/omni.drivesim.sensors.nv.radar' or '/isaac-sim/exts/omni.drivesim.sensors.nv.radar/config'
[0.133s] [ext: omni.stats-0.0.0] startup
[0.164s] [ext: omni.rtx.shadercache-1.0.0] startup
[0.170s] [ext: omni.assets.plugins-0.0.0] startup
[0.172s] [ext: omni.gpu_foundation-0.0.0] startup
2024-01-21 13:16:43 [164ms] [Warning] [carb] FrameworkImpl::setDefaultPlugin(client: omni.gpu_foundation_factory.plugin, desc : [carb::graphics::Graphics v2.11], plugin : carb.graphics-vulkan.plugin) failed. Plugin selection is locked, because the interface was previously acquired by: 
[0.182s] [ext: carb.windowing.plugins-1.0.0] startup
2024-01-21 13:16:43 [169ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2024-01-21 13:16:43 [170ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.3]) (impl: carb.windowing-glfw.plugin)
[0.183s] [ext: omni.kit.renderer.init-0.0.0] startup
2024-01-21 13:16:43 [236ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
[... cut ...]
Here is a list of available scene types:

[1] InteractiveTraversableScene: Procedurally generated scene with fully interactive objects

Choose a scene type (enter a number from 1 to 1): 

...
```

The error seems to be:
```
2024-01-21 13:16:43 [236ms] [Error] [carb.glinterop.plugin] GLInteropContext::init: carb::windowing is not available
```
And there are a few GLFW-related warning messages before. 

How to deal with this? Does this mean the docker image doesn't have ll it's needed to run OmniGibson? 

Help appreciated!

cgokmen, 2024-01-21:
Hi, how are you running the docker container? Make sure you are running it using our run_docker script and *without* the -h argument that runs everything in headless mode. Also keep an eye out for an error that might pop up when you run the run_docker script: if there's anything related to xhost please paste it here so we can help debug. What's your OS / window manager / etc?

cgokmen, 2024-01-21:
Also if the issue is not removing the -h argument, feel free to join our discord channel using the link on the repo's homepage - you can just post on the support channel there and we can stay in touch a bit faster.

ChengshuLi, 2024-07-16:
Closing the issue for now, feel free to reopen if there are further questions!

closed 2024-07-16 by ChengshuLi