StanfordVL/BEHAVIOR-1K #683: Error about no such file or directory in the /tmp folder

StoneT2000, 2024-03-22:
Thanks for making a new release of this work!

I tried running the code but had a few issues. The first is here, upon running the scene tour demo and picking the school environment and not loading the interactive objects, I get

```
At:
  /home/stao/mambaforge/envs/omnigibson/lib/python3.10/shutil.py(715): rmtree
  /home/stao/work/external/OmniGibson/omnigibson/__init__.py(54): cleanup
  /home/stao/work/external/OmniGibson/omnigibson/__init__.py(64): shutdown
  /home/stao/work/external/OmniGibson/omnigibson/__init__.py(75): shutdown_handler
  /home/stao/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/extscore/omni.kit.async_engine/omni/kit/async_engine/async_engine.py(174): _on_update
  /home/stao/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py(701): step
  /home/stao/work/external/OmniGibson/omnigibson/simulator.py(777): step
  /home/stao/work/external/OmniGibson/omnigibson/envs/env_base.py(515): step
  /home/stao/work/external/OmniGibson/omnigibson/examples/scenes/scene_tour_demo.py(104): main
  /home/stao/work/external/OmniGibson/omnigibson/examples/scenes/scene_tour_demo.py(107): <module>
  /home/stao/mambaforge/envs/omnigibson/lib/python3.10/runpy.py(86): _run_code
  /home/stao/mambaforge/envs/omnigibson/lib/python3.10/runpy.py(196): _run_module_as_main

2024-03-22 18:56:41 [78,403ms] [Error] [carb.events.python] FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmpn2o3kz6g
```
This happens whenever I try to close the viewer window and now the viewer window won't close. Any idea?

cgokmen, 2024-03-26:
This shutdown behavior is indeed a bit problematic & we are working on fixing it. For now, please exit by pressing the Esc key while the viewport is in focus.

QiSu77, 2024-10-17:
Since I press the Esc key,  I still can not close the viewer window