StanfordVL/BEHAVIOR-1K #687: [omni.physx.plugin] error running Behavior 1K tasks

elated-sawyer, 2024-03-25:
I'm trying to execute the example task in the Behavior 1K task: 

`python -m omnigibson.examples.scenes.scene_selector`

and I'm encountering the following error message related to [omni.physx.plugin]:

> 2024-03-25 22:34:33 [57,013ms] [Error] [omni.physx.plugin] Transformation change on non-root links is not supported. Link: /World/robot0/scan_link
2024-03-25 22:34:33 [57,013ms] [Error] [omni.physx.plugin] Transformation change on non-root links is not supported. Link: /World/robot0_dummy/scan_link
2024-03-25 22:34:33 [57,013ms] [Error] [omni.physx.plugin] Transformation change on non-root links is not supported. Link: /World/robot0/gyro_link
2024-03-25 22:34:33 [57,013ms] [Error] [omni.physx.plugin] Transformation change on non-root links is not supported. Link: /World/robot0/cliff_sensor_right_link
2024-03-25 22:34:33 [57,013ms] [Error] [omni.physx.plugin] Transformation change on non-root links is not supported. Link: /World/robot0/cliff_sensor_left_link
2024-03-25 22:34:33 [57,013ms] [Error] [omni.physx.plugin] Transformation change on non-root links is not supported. Link: /World/robot0/cliff_sensor_front_link
2024-03-25 22:34:33 [57,013ms] [Error] [omni.physx.plugin] Transformation change on non-root links is not supported. Link: /World/robot0/camera_rgb_optical_frame
2024-03-25 22:34:33 [57,013ms] [Error] [omni.physx.plugin] Transformation change on non-root links is not supported. Link: /World/robot0/eyes
2024-03-25 22:34:33 [57,013ms] [Error] [omni.physx.plugin] Transformation change on non-root links is not supported. Link: /World/robot0/camera_depth_optical_frame
......
2024-03-25 22:34:33 [57,047ms] [Warning] [carb] Client omni.physx.tensors.plugin has acquired [omni::physx::IPhysxSimulation v1.1] 100 times. Consider accessing this interface with carb::getCachedInterface() (Performance warning)
[WARNING] [omnigibson.utils.registry_utils] Object 'robot0' is not in the state dict to load from. Skip loading its state.
2024-03-25 22:34:33 [57,127ms] [Warning] [omnigibson.utils.registry_utils] Object 'robot0' is not in the state dict to load from. Skip loading its state.

I got to this interface:
[image]

then the OmniGibson shuts down in the progress of Resetting environment

> [INFO] [omnigibson] Resetting environment
[INFO] [omnigibson] Resetting environment
[INFO] [omnigibson] Resetting environment
2024-03-25 22:35:24 [107,299ms] [Warning] [carb] Client omni.ui has acquired [omni::ui::IGlyphManager v1.0] 100 times. Consider accessing this interface with carb::getCachedInterface() (Performance warning)
[INFO] [omnigibson] Resetting environment
[INFO] [omnigibson] ---------- Shutting Down OmniGibson ----------
2024-03-25 22:35:30 [113,704ms] [Warning] [carb] Plugin interface for a client: omni.hydratexture.plugin was already released.
2024-03-25 22:35:30 [113,704ms] [Warning] [carb] Plugin interface for a client: omni.hydratexture.plugin was already released.
2024-03-25 22:35:30 [113,705ms] [Warning] [carb] Plugin interface for a client: omni.hydratexture.plugin was already released.
2024-03-25 22:35:30 [113,705ms] [Warning] [omni.usd] Unexpected reference count of 2 for UsdStage 'anon:0x27b66270:World2.usd' while being closed in UsdContext (this may indicate it is still resident in memory).
[113.787s] Simulation App Shutting Down
2024-03-25 22:35:30 [113,786ms] [Warning] [carb] [Plugin: omni.spectree.delegate.plugin] Module /home/sawyer/Omniverse/Library/isaac_sim-2023.1.1/kit/exts/omni.usd_resolver/bin/libomni.spectree.delegate.plugin.so remained loaded after unload request
2024-03-25 22:35:30 [113,838ms] [Warning] [omni.core.ITypeFactory] Module /home/sawyer/Omniverse/Library/isaac_sim-2023.1.1/kit/exts/omni.activity.core/bin/libomni.activity.core.plugin.so remained loaded after unload request.

How to solve this problem please? Thanks for the help!

elated-sawyer, 2024-03-26:
According to https://docs.omniverse.nvidia.com/isaacsim/latest/known_issues.html, it looks like this is OK. The program works fine with the following commands `python -m omnigibson.examples.robots.robot_control_example`.

I will close this issue.

closed 2024-03-26 by elated-sawyer

cgokmen, 2024-03-26:
Hi, the `[Error] [omni.physx.plugin] Transformation change on non-root links is not supported` errors are benign, we will be silencing them in our next release. As for shutting down, that example shuts down after a few resets (take a look at the source code). Thanks!

elated-sawyer, 2024-03-26:
> Hi, the `[Error] [omni.physx.plugin] Transformation change on non-root links is not supported` errors are benign, we will be silencing them in our next release. As for shutting down, that example shuts down after a few resets (take a look at the source code). Thanks!

Thanks for the reply, got it!

RookieXwc, 2024-09-18:
@cgokmen Hello, I have encountered the same issue. When will the new release plan that can solve this bug be launched?

ChengshuLi, 2024-09-23:
@RookieXwc Replied to you in https://github.com/StanfordVL/OmniGibson/issues/892.

xiaoQ0077, 2024-11-18:
> Hi, the errors are benign, we will be silencing them in our next release. As for shutting down, that example shuts down after a few resets (take a look at the source code). Thanks!`[Error] [omni.physx.plugin] Transformation change on non-root links is not supported`
Hello! Is there any solution to this warning: **[Warning] [carb] Client omni.ui has acquired [omni::ui::IGlyphManager v1.0] 100 times. Consider accessing this interface with carb::getCachedInterface() (Performance warning)**? Because I found that after this warning appeared, the robot got stuck, and then another warning was given: **[Warning] [omni.kit.window.property.templates.simple_property_widget]** , and it has been stagnant.