StanfordVL/BEHAVIOR-1K #690: Unable to get_obs() of cam in examples.objects.draw_bounding_box

HATAEWOOK, 2024-03-27:
I got error message like this at line 69: **obs, _ = cam.get_obs()**

`Traceback (most recent call last):
  File "{USER_PATH}\anaconda3\envs\omnigibson\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "{USER_PATH}\anaconda3\envs\omnigibson\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "{USER_PATH}\Desktop\OmniGibson\omnigibson\examples\objects\draw_bounding_box.py", line 87, in <module>
    main()
  File "{USER_PATH}\Desktop\OmniGibson\omnigibson\examples\objects\draw_bounding_box.py", line 69, in main
    obs, _ = cam.get_obs()
  File "{USER_PATH}\Desktop\OmniGibson\omnigibson\sensors\sensor_base.py", line 76, in get_obs
    obs, info = self._get_obs()
  File "{USER_PATH}\Desktop\OmniGibson\omnigibson\sensors\vision_sensor.py", line 284, in _get_obs
    obs[modality] = self._remap_bounding_box_semantic_ids(obs[modality])
  File "{USER_PATH}\Desktop\OmniGibson\omnigibson\sensors\vision_sensor.py", line 411, in _remap_bounding_box_semantic_ids
    bbox["semanticId"] = VisionSensor.SEMANTIC_REMAPPER.remap_bbox(bbox["semanticId"])
  File "{USER_PATH}\Desktop\OmniGibson\omnigibson\utils\vision_utils.py", line 140, in remap_bbox
    assert semantic_id < len(self.key_array), f"Semantic id {semantic_id} is out of range!"
AssertionError: Semantic id 0 is out of range!`

It seems like 'key\_array' in 'vision\_utils.py' is empty. so they cannot remap bbox. 
Any solutions for this?

cgokmen, 2024-03-27:
We are aware of this and will push a fix soon in a hotfix release. Meanwhile can you try the og-develop branch to see if our fix helps?

cgokmen, 2024-03-27:
Cc @hang-yin

HATAEWOOK, 2024-03-27:
It works well at og-develop branch! Thanks a lot!

closed 2024-03-27 by cgokmen