StanfordVL/BEHAVIOR-1K #716: Teleoperation using Game Controller

vantony1, 2024-04-28:
Hi,

I was wondering if tele-operating robots in OmniGibson using a Game Controller is possible? I have looked that the documentation and the code but can't seem to find the keyboard teleoperation file that I could use as a guide to implement the game controller teleop functionality if one is not provided already.

Best,

Victor

cgokmen, 2024-04-28:
@wensi-ai

wensi-ai, 2024-04-29:
Yes, using game controller is possible. Take a look at [this script](https://github.com/StanfordVL/OmniGibson/blob/main/omnigibson/utils/teleop_utils.py) where we use [Telemoma](https://github.com/UT-Austin-RobIn/telemoma) for robot teleoperation (you can try out `robot_teleoerate_demo.py` to see how it is used)

Alternatively, [here](https://github.com/StanfordVL/OmniGibson/blob/main/omnigibson/utils/ui_utils.py) is another keyboard interface that is used by `robot_control_example.py`

vantony1, 2024-04-29:
Thanks so much! I will give it a shot and let you know if I further questions. I appreciate the help.

closed 2024-05-17 by wensi-ai