StanfordVL/BEHAVIOR-1K #721: Saving and Loading Simulation State Error

Arpitrf, 2024-05-10:
Hello, I am trying to save and load the simulation state by following this: [https://behavior.stanford.edu/omnigibson/tutorials/demo_collection.html#optional-saving-and-loading-simulation-state](https://behavior.stanford.edu/omnigibson/tutorials/demo_collection.html#optional-saving-and-loading-simulation-state)

I saved the simulation state as follows: 
```
og.sim.save(f'sim_state.json')
```
And then I am loading the state as follows:
```
# Load the environment 
env = og.Environment(configs=config)
og.sim.restore('sim_state.json')
```
However, I get the following error:
```
Traceback (most recent call last):
  File "/code/omnigibson/envs/env_base.py", line 524, in step
    self._scene_graph_builder.step(self.scene)
  File "/code/omnigibson/scene_graphs/graph_builder.py", line 146, in step
    desired_frame_to_world, world_to_desired_frame = self._get_desired_frame()
  File "/code/omnigibson/scene_graphs/graph_builder.py", line 63, in _get_desired_frame
    desired_frame_to_world = self._get_robot_to_world_transform()
  File "/code/omnigibson/scene_graphs/graph_builder.py", line 69, in _get_robot_to_world_transform
    robot_to_world = self._robot.get_position_orientation()
  File "/code/omnigibson/robots/tiago.py", line 697, in get_position_orientation
    return self.base_footprint_link.get_position_orientation()
  File "/code/omnigibson/prims/rigid_prim.py", line 314, in get_position_orientation
    pos, ori = self._rigid_prim_view.get_world_poses()
  File "/code/omnigibson/prims/rigid_prim.py", line 355, in _rigid_prim_view
    assert self._rigid_prim_view_direct.is_physics_handle_valid() and \
AssertionError: Rigid prim view must be valid if physics is running!
``` 
Request your help!

wensi-ai, 2024-05-10:
@cremebrule

wensi-ai, 2024-05-17:
Solved by changing `og.sim.restore('sim_state.json')` to `config["scene"]['scene_file'] = 'sim_state.json'`. Also removing the scene graph part from the config file could solve the problem.