StanfordVL/BEHAVIOR-1K #727: Issues running behavior example: `python -m omnigibson.examples.environments.behavior_env_demo`

AshisGhosh, 2024-05-17:
Hi, I'm running into the following error when running
```
python -m omnigibson.examples.environments.behavior_env_demo
```

```
2024-05-17 22:51:25 [39,334ms] [Error] [omni.physx.plugin] Particles feature is only supported on GPU. Please enable GPU dynamics flag in Property/Scene of physics scene!
2024-05-17 22:51:26 [39,841ms] [Warning] [omni.syntheticdata.plugin] SdRenderVarPtr missing valid input renderVar DistanceToCameraSDhost
[WARNING] [omnigibson.utils.registry_utils] Object 'water' is not in the state dict to load from. Skip loading its state.
2024-05-17 22:51:26 [40,179ms] [Warning] [omnigibson.utils.registry_utils] Object 'water' is not in the state dict to load from. Skip loading its state.
[WARNING] [omnigibson.objects.stateful_object] Missing object state [AttachedTo] in the state dump for obj mirror_pevdqu_0
2024-05-17 22:51:26 [40,194ms] [Warning] [omnigibson.objects.stateful_object] Missing object state [AttachedTo] in the state dump for obj mirror_pevdqu_0
[WARNING] [omnigibson.objects.stateful_object] Missing object state [ParticleSource] in the state dump for obj sink_ksecxq_0
2024-05-17 22:51:26 [40,196ms] [Warning] [omnigibson.objects.stateful_object] Missing object state [ParticleSource] in the state dump for obj sink_ksecxq_0
[WARNING] [omnigibson.objects.stateful_object] Missing object state [ParticleSource] in the state dump for obj sink_zexzrc_0
2024-05-17 22:51:26 [40,197ms] [Warning] [omnigibson.objects.stateful_object] Missing object state [ParticleSource] in the state dump for obj sink_zexzrc_0
[WARNING] [omnigibson.utils.registry_utils] Object 'robot0' is not in the state dict to load from. Skip loading its state.
2024-05-17 22:51:26 [40,205ms] [Warning] [omnigibson.utils.registry_utils] Object 'robot0' is not in the state dict to load from. Skip loading its state.
[INFO] [omnigibson.utils.bddl_utils] Sampling task...
Traceback (most recent call last):
  File "/omnigibson-src/omnigibson/examples/environments/behavior_env_demo.py", line 58, in <module>
    main()
  File "/omnigibson-src/omnigibson/examples/environments/behavior_env_demo.py", line 36, in main
    env = og.Environment(configs=cfg)
  File "/omnigibson-src/omnigibson/utils/python_utils.py", line 90, in wrapper
    func(*values.args, **values.kwargs)
  File "/omnigibson-src/omnigibson/envs/env_base.py", line 77, in __init__
    self.load()
  File "/omnigibson-src/omnigibson/envs/env_base.py", line 373, in load
    self._load_task()
  File "/omnigibson-src/omnigibson/envs/env_base.py", line 182, in _load_task
    self._task.load(env=self)
  File "/omnigibson-src/omnigibson/tasks/task_base.py", line 111, in load
    self._load(env=env)
  File "/omnigibson-src/omnigibson/tasks/behavior_task.py", line 190, in _load
    success, self.feedback = self.initialize_activity(env=env)
  File "/omnigibson-src/omnigibson/tasks/behavior_task.py", line 309, in initialize_activity
    accept_scene, feedback = self.sampler.sample()
  File "/omnigibson-src/omnigibson/utils/bddl_utils.py", line 577, in sample
    accept_scene, feedback = self._prepare_scene_for_sampling()
  File "/omnigibson-src/omnigibson/utils/bddl_utils.py", line 649, in _prepare_scene_for_sampling
    error_msg = self._build_inroom_object_scope()
  File "/omnigibson-src/omnigibson/utils/bddl_utils.py", line 872, in _build_inroom_object_scope
    valid_models = {cat: set(get_all_object_category_models_with_abilities(
  File "/omnigibson-src/omnigibson/utils/bddl_utils.py", line 872, in <dictcomp>
    valid_models = {cat: set(get_all_object_category_models_with_abilities(
  File "/omnigibson-src/omnigibson/utils/asset_utils.py", line 283, in get_all_object_category_models_with_abilities
    if supports_abilities(abilities_info, prim):
  File "/omnigibson-src/omnigibson/utils/asset_utils.py", line 266, in supports_abilities
    if not requirement.is_compatible_asset(prim=obj_prim)[0]:
  File "/omnigibson-src/omnigibson/object_states/object_state_base.py", line 53, in is_compatible_asset
    raise NotImplementedError
NotImplementedError
```

##Environment:
Running locally with docker.

Any advice?

cgokmen, 2024-05-17:
Hi, can you use the offline mode of this example please? The online sampling feature does not work on all tasks currently. @hang-yin can you please confirm that this example works currently?

hang-yin, 2024-05-18:
@cgokmen can confirm that this works without online sampling. 
Hi @AshisGhosh, could you try selecting 1 when prompted?

AshisGhosh, 2024-05-18:
Hi! Thanks for the quick responses. Unfortunately that does not work for me either and I get a different (CUDA related) error. Note that I am able to run all the other examples I have tried so far without problem (teleop, action primitives, etc)

```
2024-05-18 19:21:00 [78,579ms] [Error] [carb.cudainterop.plugin] CUDA error 700: cudaErrorIllegalAddress - an illegal memory access was encountered)
2024-05-18 19:21:00 [78,579ms] [Error] [carb.cudainterop.plugin] Failed to wait on external semaphore in CUDA.
2024-05-18 19:21:00 [78,579ms] [Error] [gpu.foundation.plugin] Wait for external semaphore failed in CUDA submission in command list "Render graph command list (Render queue 0, device 0, frame submission index 1)"
2024-05-18 19:21:00 [78,588ms] [Error] [carb.cudainterop.plugin] CUDA error 700: cudaErrorIllegalAddress - an illegal memory access was encountered)
2024-05-18 19:21:00 [78,588ms] [Error] [carb.cudainterop.plugin] Failed to free mipmapped array in CUDA.
2024-05-18 19:21:00 [78,588ms] [Error] [carb.cudainterop.plugin] CUDA error 700: cudaErrorIllegalAddress - an illegal memory access was encountered)
2024-05-18 19:21:00 [78,588ms] [Error] [carb.cudainterop.plugin] Failed to free mipmapped array in CUDA.
2024-05-18 19:21:00 [78,626ms] [Error] [carb.cudainterop.plugin] CUDA error 700: cudaErrorIllegalAddress - an illegal memory access was encountered)
2024-05-18 19:21:00 [78,626ms] [Error] [carb.cudainterop.plugin] Failed to wait on external semaphore in CUDA.
2024-05-18 19:21:00 [78,626ms] [Error] [gpu.foundation.plugin] Wait for external semaphore failed in CUDA submission in command list "Render graph command list (Render queue 0, device 0, frame submission index 2)"
Traceback (most recent call last):
  File "/micromamba/envs/omnigibson/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/micromamba/envs/omnigibson/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/omnigibson-src/omnigibson/examples/environments/behavior_env_demo.py", line 58, in <module>
    main()
  File "/omnigibson-src/omnigibson/examples/environments/behavior_env_demo.py", line 36, in main
    env = og.Environment(configs=cfg)
  File "/omnigibson-src/omnigibson/utils/python_utils.py", line 90, in wrapper
    func(*values.args, **values.kwargs)
  File "/omnigibson-src/omnigibson/envs/env_base.py", line 77, in __init__
    self.load()
  File "/omnigibson-src/omnigibson/envs/env_base.py", line 370, in load
    self._load_scene()
  File "/omnigibson-src/omnigibson/envs/env_base.py", line 203, in _load_scene
    og.sim.import_scene(scene)
  File "/omnigibson-src/omnigibson/simulator.py", line 466, in import_scene
    self._scene.initialize()
  File "/omnigibson-src/omnigibson/scenes/scene_base.py", line 339, in initialize
    og.sim.load_state(init_state, serialized=False)
  File "/omnigibson-src/omnigibson/simulator.py", line 1371, in load_state
    assert self.is_playing()
AssertionError

```

ChengshuLi, 2024-07-16:
I am unable to reproduce this error on my end. Closing the issue for now. Feel free to re-open if this is still an issue.

closed 2024-07-16 by ChengshuLi