StanfordVL/BEHAVIOR-1K #733: examples/observations does not exist

ZZWENG, 2024-05-23:
Hi, 
The README under `examples` folder says that there is a `observations` folder. But it does not exist. Could you point me to where it is?

Also, is there any demos that shows how to render and save a video of the scene on a headless server?

wensi-ai, 2024-05-23:
Hi, apologies for the confusion, the README under examples hasn't been updated for a while. There is actually no observation folder. 

For rendering sensors you can take a look at [the sensor module documentation](https://behavior.stanford.edu/omnigibson/modules/sensor.html). Basically you can create a `VisionSensor` and get the rendered image with `get_obs()` at each timestep.

ChengshuLi, 2024-07-16:
Apologies! That README is completely outdated. Will remove it in the next release.

Regarding your question about how to render and save a video of the scene on a headless sensor, you can also use the default viewer camera.

```
# By default, images are of shape (720, 1280, 4)
img1 = og.sim.viewer_camera.get_obs()[0]["rgb"]
og.sim.viewer_camera.set_position_orientation(new_pos, new_orn)
og.sim.render()
img2 = og.sim.viewer_camera.get_obs()[0]["rgb"]
...
# Save img1, img2, ... into a video.
```

closed 2024-07-16 by ChengshuLi