StanfordVL/BEHAVIOR-1K #744: Camera orientation mismatch in scaling robot size

MhdMohammadi, 2024-06-04:
I'm trying to import "fetch" into one of the scenes, and because the robot's size is relatively too big, I want to rescale it to avoid possible collisions. However, it looks like the camera orientation cannot be set correctly.
```
File "/omnigibson-src/omnigibson/prims/xform_prim.py", line 141, in _set_xform_properties
    assert np.allclose(new_position, current_position, atol=1e-4) and np.allclose(r1, r2, atol=1e-4), \
AssertionError: /World/robot0/eyes/Camera: old_pos: [1.67729348e-01 5.24871461e-17 1.04592872e+00], new_pos: [1.67729348e-01 5.24871477e-17 1.04592872e+00], old_orn: [ 0.43045944 -0.43045944 -0.56098545  0.56098545], new_orn: [-0.47755187  0.20257453  0.62235748 -0.58615608]
```

And this part of the code is asserting the mismatch between the new and old orientation values.

```
self.set_position_orientation(position=current_position, orientation=current_orientation)
new_position, new_orientation = self.get_position_orientation()
r1 = R.from_quat(current_orientation).as_matrix()
r2 = R.from_quat(new_orientation).as_matrix()
# Make sure setting is done correctly
assert np.allclose(new_position, current_position, atol=1e-4) and np.allclose(r1, r2, atol=1e-4), \
    f"{self.prim_path}: old_pos: {current_position}, new_pos: {new_position}, " \
    f"old_orn: {current_orientation}, new_orn: {new_orientation}"
```

And this is the config file to reproduce the settings.

```
env:
    action_frequency: 60                  # (int): environment executes action at the action_frequency rate
    physics_frequency: 60                 # (int): physics frequency (1 / physics_timestep for physx)
    # device: null                          # (None or str): specifies the device to be used if running on the gpu with torch backend
    automatic_reset: false                # (bool): whether to automatic reset after an episode finishes
    flatten_action_space: false           # (bool): whether to flatten the action space as a sinle 1D-array
    flatten_obs_space: false              # (bool): whether the observation space should be flattened when generated
    use_external_obs: false               # (bool): Whether to use external observations or not
    initial_pos_z_offset: 0.1
    external_sensors: null

scene:
    type: InteractiveTraversableScene
    scene_model: Benevolence_1_int
    trav_map_resolution: 0.1
    default_erosion_radius: 0.0
    trav_map_with_objects: true
    num_waypoints: 1
    waypoint_resolution: 0.2
    load_room_types: null
    load_room_instances: null
    load_task_relevant_only: false
    seg_map_resolution: 0.1
    scene_source: OG
    include_robots: true

robots:
    - type: Fetch
      obs_modalities: [proprioception, scan, rgb, depth_linear, seg_semantic, seg_instance]
      scale: 0.5
      action_type: continuous
      action_normalize: True
      visual_only: true

task:
    type: DummyTask
```

hang-yin, 2024-07-01:
Hi @MhdMohammadi, sorry for taking so long to get back to you. Unfortunately, we do not currently support scaling robots. If you want to avoid collisions in a scene, here are some alternatives you could try: 

1. Try a scene with more empty space. You can check here for a list of scene "floor plans": https://behavior.stanford.edu/omnigibson/modules/scene.html
2. Only load a subset of objects into your scene. e.g. `cfg["scene"]["load_object_categories"] = ["floors", "walls", "ceilings"]`

I'm happy to chat more if you need anything else!

MhdMohammadi, 2024-07-31:
Thank you, @hang-yin, for your reply. It's unfortunate, especially since we had this feature in iGibson. Many scenes, mostly apartments, have narrow corridors, and the default size of the robot makes it almost impossible to complete tasks.

hang-yin, 2024-07-31:
Hi @MhdMohammadi , sorry for the trouble! When I was looking into this a month ago, I tried to modify our infra to support this but it turned out to be a non-trivial effort. If this turns out to be a necessary feature to support, I can definitely take a look again once I'm done with what I'm currently working on. Here's one other thing you could try: if the robot arm is the problem when navigating around narrow spaces, try `robot.tuck()` so that the arm don't get into your way.

MhdMohammadi, 2024-08-01:
Hi @hang-yin ,

Thanks a lot for your quick response! I'll take a look into `robot.tuck()`. It might help; however, the body itself is also too large, and it won't solve the problem completely. It would definitely be great if we could have this feature.