StanfordVL/BEHAVIOR-1K #782: OmniGibson does not contain all objects from BEHAVIOR-1K Objects

vts3rss, 2024-06-29:
There are objects on the [Behavior-1K Objects site](https://behavior.stanford.edu/knowledgebase/objects/index.html) that are not available on OmniGibson. By my count, about 500 objects are missing.  

How to download the full version of the dataset with missing objects?

I found the dataset archive in the source code. https://storage.googleapis.com/gibson_scenes/og_dataset_1_0_0.tar.gz 
Perhaps there is a link to the updated archive?

Missing objects full list:
[missing_objects_full_list.json](https://github.com/user-attachments/files/16039278/missing_objects_full_list.json)

Some of the missing objects:

alga-acjxoq
https://behavior.stanford.edu/knowledgebase/objects/alga-acjxoq/index.html

almond-acooui
https://behavior.stanford.edu/knowledgebase/objects/almond-acooui/index.html

bird_feed-ozvqdx
https://behavior.stanford.edu/knowledgebase/objects/bird_feed-ozvqdx/index.html

black_bean-ncvtfs
https://behavior.stanford.edu/knowledgebase/objects/black_bean-ncvtfs/index.html

blueberry-bajppk
https://behavior.stanford.edu/knowledgebase/objects/blueberry-bajppk/index.html

bottle_of_tomato_juice-csumos
https://behavior.stanford.edu/knowledgebase/objects/bottle_of_tomato_juice-csumos/index.html

box_of_wagashi-mlmrwy
https://behavior.stanford.edu/knowledgebase/objects/box_of_wagashi-mlmrwy/index.html

breadcrumb-fjdkah
https://behavior.stanford.edu/knowledgebase/objects/breadcrumb-fjdkah/index.html

bunchgrass-kqhokv
https://behavior.stanford.edu/knowledgebase/objects/bunchgrass-kqhokv/index.html

cashew-qyglnm
https://behavior.stanford.edu/knowledgebase/objects/cashew-qyglnm/index.html

cgokmen, 2024-06-29:
Most (all?) of those are substances - the objects form the individual particles. You can find the meshes in the systems directory of the dataset. They can be sampled using the particlesystem APIs in OG.

vts3rss, 2024-06-29:
> Most (all?) of those are substances - the objects form the individual particles. You can find the meshes in the systems directory of the dataset. They can be sampled using the particlesystem APIs in OG.

Indeed, I found most of them in the systems folder! There are 96 unfound objects left. Maybe they are also stored in some folder? I have attached json with their names.

[missing_objects_new.json](https://github.com/user-attachments/files/16041591/missing_objects_new.json)

vts3rss, 2024-06-29:
@cgokmen Slightly unrelated question. Many objects look like low-poly. Most likely to optimize computing resources. Is it possible to get a high-poly version of a dataset or individual objects? Thank you in advance!

cgokmen, 2024-06-30:
We do hope to include multiple qualities of the assets in the future but this is not possible currently. As for the missing objects, they will be included in an upcoming release of the dataset as certain issues are fixed.

closed 2024-06-30 by cgokmen

vts3rss, 2024-07-01:
Thank you very much!