StanfordVL/BEHAVIOR-1K #787: Segmation fault - og-develop branch Isaacsim 4.0.0

TuluhanAkbulut, 2024-07-08:
Hello,

I was trying to robot control example using Isaacsim 4.0.0 but I am getting the following error:

2024-07-08 22:42:23 [9,354ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2024-07-08 22:42:23 [9,480ms] [Warning] [carb] Client omni.syntheticdata.plugin Failed to acquire interface [omni::graph::core::INode v4.10] while unloading all plugins
Fatal Python error: Segmentation fault

Also most objects are not loaded because:
[WARNING] [omnigibson.utils.registry_utils] Object 'cooked__diced__pumpkin' is not in the state dict to load from. Skip loading its state.

I am planning to use Panda arm in a kitchen environment. Is there any stable commits that can run with 4.0.0 that I can revert to?

cgokmen, 2024-07-08:
Hi,

Can you try running the og-develop branch (just recently introduced 4.0.0 compatibility, this is our development branch)? Also please post the full log - the replicator and state loading errors are safe to ignore, so the segfault must be caused by something else.

Thanks!

TuluhanAkbulut, 2024-07-09:
I was using og-development branch actually. 
[err.txt](https://github.com/user-attachments/files/16136649/err.txt)

Thank you!

cgokmen, 2024-07-16:
This is an issue on the current og-develop branch related to a new API we introduced recently. Fix for it should be merged later today / tomorrow.

closed 2024-08-21 by cgokmen