StanfordVL/BEHAVIOR-1K #800: Issues about multi-agent settings

yubinwang11, 2024-07-13:
It seems that the function of _scene_graph_builder.start(self.scene) in OmniGibson/omnigibson/scene_graphs/graph_builder.py only supports single agent settings:
`        if self._robot_name is None:
            assert (
                len(scene.robots) == 1
            ), "Cannot build scene graph without specifying robot name if there are multiple robots."
            self._robot = scene.robots[0]
        else:
            self._robot = scene.object_registry("name", self._robot_name)
            assert self._robot, f"Robot with name {self._robot_name} not found in scene."
`
The self._robot_name should be a list instead of a string when multi-agent is considered.

Thanks a lot for your review.

cgokmen, 2024-07-16:
Thanks for the report. We will fix this for our next release.

hang-yin, 2024-08-15:
Here's a proposed fix: https://github.com/StanfordVL/OmniGibson/pull/836

closed 2024-08-21 by hang-yin